FreeCAD/FreeCAD #14587: Flattened SVG export issues

davesrocketshop, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The attached project file has several SVG export issues.

This is the piece modeled using Part Workbench:
[image]

Selecting the part and exporting as a Flattened SVG produces this when imported into Inkscape:
[image]

If you then ungroup the objects you get this:
[image]

Ungrouping the objects again, you can hit delete twice removing duplicate layers and finally winding up with a single copy:
[image]

[SVG Error.FCStd.zip](https://github.com/user-attachments/files/15768823/SVG.Error.FCStd.zip)

### Full version info

```shell
Dev version built from source:

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37670 (Git)
Build type: Release
Branch: main
Hash: ba0ff0db5e78973cb994b82f27bdd228c7ad41bd
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ArchTextures
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * fcgear 1.0.0
  * FeedsAndSpeeds 0.5.0
  * GDML.backup1701064072.5104816 2.0.0 (Disabled)
  * kicadStepUpMod 11.1.2
  * LCInterlocking
  * MaterialTest
  * OpenTheme 2024.5.3
  * render 2023.8.18
  * Rocket 4.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-11:
Maybe related https://github.com/FreeCAD/FreeCAD/issues/13249

Roy-043, 2024-06-11:
I get a different result. See attached SVG. There is a filled outline, and each edge of that shape is duplicated. Which does not seem to match the description. Note that I have only selected the End body for export.

[image]

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods: 
```

mnesarco, 2025-02-18:
@davesrocketshop  I have opened the file SVG.Error.FCStd in FreeCAD 1.0, selected End and exported to svg using SvgWorkbench for testing, this is the result:

[image]

In SvgWorkbench I use TechDraw+DraftWB APIs to project the svg geometry. So if there are bugs there they will leak to SvgWorkbench too.

Roy-043, 2025-02-28:
Looking at this again I have to correct what I said before: There are 2 filled outlines and every edge is duplicated. And, as @tigert has mentioned, each edge is in fact a closed path (with duplicate points at each end).

The reason for this result is that each face of the solid gets exported, incuding faces that do not face the camera.

Roy-043, 2025-07-03:
Still an issue:
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42340 (Git)
Build date: 2025/06/27 09:25:36
Build type: Release
Branch: main
Hash: 4f167b828b68b1544732881f0b2393ea873cffd1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

luzpaz, 2025-11-29:
* Will #24832 impact this ?

@davesrocketshop any chance you can test ?

davesrocketshop, 2025-11-30:
> * Will [Draft: Fix duplicated curves in SVG export and add sub-selection support #24832](https://github.com/FreeCAD/FreeCAD/pull/24832) impact this ?
> 
> 
> [@davesrocketshop](https://github.com/davesrocketshop) any chance you can test ?

I'll test it today