FreeCAD/FreeCAD #14588: [Part Design] Pocket operation introducing strange artifact

davesrocketshop, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The included file introduces a strange artifact when performing a pocket operation (the last operation in the file). I have a shape that I then want to pocket. Apply a sketch of a circle going through all and a strange artifact appears.
[image]
[image]

This is not an issue with the drawing. Exporting as a .3mf (or any other format) shows the artifact is part of the object.
[image]

[SVG Error.FCStd.zip](https://github.com/user-attachments/files/15771914/SVG.Error.FCStd.zip)

### Full version info

```shell
Latest Dev version built from source:

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37670 (Git)
Build type: Release
Branch: main
Hash: ba0ff0db5e78973cb994b82f27bdd228c7ad41bd
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ArchTextures
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * fcgear 1.0.0
  * FeedsAndSpeeds 0.5.0
  * GDML.backup1701064072.5104816 2.0.0 (Disabled)
  * kicadStepUpMod 11.1.2
  * LCInterlocking
  * MaterialTest
  * OpenTheme 2024.5.3
  * render 2023.8.18
  * Rocket 4.0.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2024-06-14:
> @davesrocketshop I found the correct file from the forum topic, here is my rework, the artifact was introduced by the first linear pattern but that wasn't really the root cause, IMHO it was the way you used a billet like you were machining it. I checked my model output in Cura and all was fine there too.
> 
> [Environmental Radiator Lower2_Syres.FCStd.zip](https://github.com/user-attachments/files/15837929/Environmental.Radiator.Lower2_Syres.FCStd.zip)
> 
> ```
> OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.37730 (Git)
> Build type: Release
> Branch: main
> Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
> Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
> Locale: English/United Kingdom (en_GB)
> Installed mods: 
>   * Silk 0.1.5
>   * freecad.gears 1.2.0
>   * Curves 0.6.23
>   * CfdOF 1.25.11
>   * fasteners 0.5.17
>   * sheetmetal 0.4.1
> ```

I have a file that works. That's not the problem. I know there are better ways to achieve the same result. The problem is that this file SHOULD work but it doesn't. In my opinion that's a bug. If there's a step in the process flow that is incorrect it should be flagged and explained. Checking geometry shows no errors when there clearly is.

davesrocketshop, 2024-06-14:
> Well IMHO this would have to be raised with OpenCascade but I'm happy to be corrected. They are quite strict on bug submissions, you can't just send them a FreeCAD file, here is an example https://tracker.dev.opencascade.org/view.php?id=31601 sent by abdullah in 2020, took nearly four years to get sorted.

You may be right but this feels like a cop out for a few reasons:
1. Is this definitively proven to be an OCC error?
2. If it is an OCC error, are there mitigations that can be done in the FC workflow?
3. Is there a way to detect and report the error that can be rolled into the current tools?
4. Is relying on the user to report upstream issues really the way to handle this?

When I asked on the forum, I got a lot of "Try it this way instead" with no answer to what was done wrong. I still don't have that answer. Yes there are more elegant ways to do it... this is an artifact from when I was learning FC during v0.16 development and I've learned a lot since then. But that doesn't change that this workflow should work and I can almost guarantee that others will try something similar.

If it is indeed an upstream problem, then expecting end users to report it upstream is unrealistic. In many cases the user won't have the familiarity with the internals required to do that. In virtually all cases, they'll walk away, perhaps from FC entirely. I have extensive internals experience and virtually no OCC experience. This needs to be properly evaluated and reported by someone with the expertise to do so. Tracking it here can also be helpful in evaluating any fixes or workarounds, as well as providing useful metrics.

maxwxyz, 2025-04-16:
@pieterhijma added OCC label