FreeCAD/FreeCAD #14593: Measurements of sketches don't save

Jookia, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a cube
2. Create a sketch
3. Import the geometry of the cube
4. Create a rectangle based on the geometry
5. Exit sketcher moder
6. Measure the length of a sketch edge
7. It will look correct:

[image]

8. Save and load the file again
9. The measurement is broken

[image]

Changing the measurement's linked object to the sketch before saving seems to cause it to save and load fine, despite having an identical name.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git)
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.12.3, Qt 6.7.1, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.67
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-11:
@hlorus FYI

maxwxyz, 2024-06-16:
@hlorus In the latest weekly I cannot measure sketches at all, the distance and length of simple sides of a rectangle is always 0

PR #14962 (2024-07-08): Measure: Fix restoring/creating measurements of sketch elements
This fixes the issues described in #14593 by by explicitly passing the last object in the subjects subobject list to getTopoShape as well as the subElementName.

fixes #14593
closed 2024-07-08 by chennes