FreeCAD/FreeCAD #14597: assembly Do not remove connected joints when deleting a component

Vincenzoferrara, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

when i remove a component from an assembly it should also remove its joints, but i get the warning

[image]

it should ask if you want to remove the joints

I use the Italian language

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT) [ OS: Italian/Italy (it_IT) ]
Installed mods: 
  * AutoOpenSpreadsheet
  * ProDarkThemePreferencePack 1.0.0
  * CommandPanel
  * IconThemes
  * fasteners.backup1715436808.37841 0.5.16 (Disabled)
  * FeedsAndSpeeds 0.5.0
  * ExplodedAssembly
  * OpenTheme 2024.5.16
  * Freecad-Built-in-themes-beta 1.2.2
  * fasteners 0.5.21
  * TabBar
  * Assembly4 0.50.12
  * CubeMenu
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-11:
This is a reasonable request.

PaddleStroke, 2024-06-11:
The error message is a general message due to the links. Here it is in english : 
[image]

I'm actually not sure how to solve this problem.
@wwmayer could you please weight in? Is it possible to trigger deletion of the related joints when a component is deleted ? I'm not sure how to handle this.

wwmayer, 2024-06-11:
It's up to your view provider class to override the method `canDelete`. If it handles the deletion it must return true.

PaddleStroke, 2024-06-11:
Thanks @wwmayer ! It works to delete the joints as same time as components by implementing `ViewProviderAssembly::canDelete`

However I still have the 'Object dependency' warning box that appears before attempting the deletion. Is there a way to prevent this message ?

wwmayer, 2024-06-11:
If you look at the code then the dialog pops up if canDelete() returns false. Which view provider class is called there?

closed 2024-06-17 by chennes