FreeCAD/FreeCAD #14598: [Problem] FreeCAD crashes on Undo (not only with fillets)

MiqCG, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When undoing an operation in Part Design (hitting Ctrl-Z or clicking on Edit>Undo), hard crashes FreeCAD. See attached file and terminal output in the "Anything else" section.

In the first video the file has fillet and the intended operation to undo is a fillet. But in the second video you can see that if you delete the fillets and the previous Pad, redo the Pad (to the opposite face) and then Ctrl-Z, FreeCAD crashes completely.

I searched on Github with [is:issue is:open crash undo] with no relevant results. Hope is not a duplicate issue.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * fasteners 0.5.22
```

### Subproject(s) affected?

PartDesign

### Anything else?

[#201-001 Barra inferior central.FCStd.zip](https://github.com/user-attachments/files/15778298/201-001.Barra.inferior.central.FCStd.zip)

https://github.com/FreeCAD/FreeCAD/assets/18265355/dabbf2cd-7d29-4dcf-90f8-7f0a1c84924e

https://github.com/FreeCAD/FreeCAD/assets/18265355/6b7815ff-c1fb-4b10-b622-70f5234744b3

Terminal output after hitting Ctrl-Z (before hitting it, there aren't any messages):

FreeCAD 0.22.0, Libs: 0.22.0devR37668 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

During initialization the error "No module named 'WebGui'" occurred in /home/miquel/.local/share/FreeCAD/Mod/FreeCAD-motion-workbench-master/InitGui.py
Please look into the log file for further information
flag
flag2
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7fd40425b050]
#1  0x7fd3a85ecd6f in PartDesignGui::ViewProviderBody::unifyVisualProperty(App::Property const*) from /tmp/.mount_FreeCARKErKQ/usr/lib/PartDesignGui.so+0x19f
#2  0x7fd3a85f369c in PartDesignGui::ViewProviderBody::onChanged(App::Property const*) from /tmp/.mount_FreeCARKErKQ/usr/lib/PartDesignGui.so+0x3c
#3  0x7fd40637a965 in App::Property::hasSetValue() from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADApp.so+0x135
#4  0x7fd3a85ee238 in PartDesignGui::ViewProviderBody::copyColorsfromTip(App::DocumentObject*) from /tmp/.mount_FreeCARKErKQ/usr/lib/PartDesignGui.so+0x6e8
#5  0x7fd3a85f33b5 in PartDesignGui::ViewProviderBody::updateData(App::Property const*) from /tmp/.mount_FreeCARKErKQ/usr/lib/PartDesignGui.so+0x1e5
#6  0x7fd407757210 in Gui::ViewProvider::update(App::Property const*) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADGui.so+0x60
#7  0x7fd40776d707 in Gui::ViewProviderDocumentObject::update(App::Property const*) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADGui.so+0x57
#8  0x7fd4072fc8fe in Gui::Document::slotChangedObject(App::DocumentObject const&, App::Property const&) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADGui.so+0x4e
#9  /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADApp.so(+0x224ce4) [0x7fd406224ce4]
#10  0x7fd406260874 in App::DocumentObject::onChanged(App::Property const*) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADApp.so+0x134
#11  0x7fd40637a965 in App::Property::hasSetValue() from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADApp.so+0x135
#12  0x7fd406321dfe in App::TransactionObject::applyChn(App::Document&, App::TransactionalObject*, bool) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADApp.so+0xae
#13  0x7fd40632209b in App::Transaction::apply(App::Document&, bool) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADApp.so+0x10b
#14  0x7fd40623594d in App::Document::undo(int) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADApp.so+0xdd
#15  0x7fd4072fa0cc in Gui::Document::undo(int) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADGui.so+0x4c
#16  0x7fd40770fcd1 in Gui::View3DInventor::onMsg(char const*, char const**) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADGui.so+0x521
#17  0x7fd40727936b in Gui::Application::sendMsgToActiveView(char const*, char const**) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADGui.so+0x3b
#18  0x7fd40738cf68 in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADGui.so+0x2a8
#19  0x7fd40738d324 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADGui.so+0x134
#20  /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7fd4046eb6f2]
#21  0x7fd40556a462 in QAction::triggered(bool) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Widgets.so.5+0x32
#22  0x7fd40556ce2b in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Widgets.so.5+0xab
#23  /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Widgets.so.5(+0x2e12e2) [0x7fd4056e12e2]
#24  /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Widgets.so.5(+0x2e8c34) [0x7fd4056e8c34]
#25  0x7fd4055aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Widgets.so.5+0x178
#26  0x7fd4055703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#27  0x7fd405578262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#28  0x7fd407338349 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADGui.so+0x99
#29  0x7fd4046b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Core.so.5+0x108
#30  0x7fd4055763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#31  /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Widgets.so.5(+0x1c72a5) [0x7fd4055c72a5]
#32  /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7fd4055ca0c9]
#33  0x7fd4055703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#34  0x7fd407338349 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADGui.so+0x99
#35  0x7fd4046b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Core.so.5+0x108
#36  0x7fd404d353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Gui.so.5+0x84c
#37  0x7fd404d0b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Gui.so.5+0xac
#38  /tmp/.mount_FreeCARKErKQ/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7fd3ff098bea]
#39  /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7fd402f10ad7]
#40  /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7fd402f13ac7]
#41  /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7fd402f140ce]
#42  0x7fd40470aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Core.so.5+0x66
#43  0x7fd4046b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Core.so.5+0x11b
#44  0x7fd4046bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libQt5Core.so.5+0x82
#45  0x7fd40729b847 in Gui::Application::runApplication() from /tmp/.mount_FreeCARKErKQ/usr/bin/../lib/libFreeCADGui.so+0xa27
#46  /tmp/.mount_FreeCARKErKQ/usr/bin/freecad(+0x7620) [0x55c02daa2620]
#47  /lib/x86_64-linux-gnu/libc.so.6(+0x2724a) [0x7fd40424624a]
#48  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7fd404246305]
#49  /tmp/.mount_FreeCARKErKQ/usr/bin/freecad(+0x7981) [0x55c02daa2981]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-06-11:
Perhaps, I'll leave the backtrace here.

<details><summary>Backtrace</summary>

```cpp
Process 1742 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
    frame #0: 0x00007fff9507d1e6 PartDesignGui.so`PartDesignGui::ViewProviderBody::unifyVisualProperty(this=0x000055555acceeb0, prop=0x000055555accf368) at ViewProviderBody.cpp:454:71
   451          }
   452 
   453          //copy over the properties data
-> 454          auto fprop = gdoc->getViewProvider(feature)->getPropertyByName(prop->getName());
   455          fprop->Paste(*prop);
   456      }
   457  }
(lldb) bt
error: libFreeCADApp.so 0x0224bde0: DW_TAG_member '_M_local_buf' refers to type 0x0000000002293443 which extends beyond the bounds of 0x0224bdd6
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
  * frame #0: 0x00007fff9507d1e6 PartDesignGui.so`PartDesignGui::ViewProviderBody::unifyVisualProperty(this=0x000055555acceeb0, prop=0x000055555accf368) at ViewProviderBody.cpp:454:71
    frame #1: 0x00007fff9507e21d PartDesignGui.so`PartDesignGui::ViewProviderBody::onChanged(this=0x000055555acceeb0, prop=0x000055555accf368) at ViewProviderBody.cpp:415:28
    frame #2: 0x00007ffff5fe2634 libFreeCADApp.so`App::Property::hasSetValue(this=0x000055555accf368) at Property.cpp:230:26
    frame #3: 0x00007ffff603c638 libFreeCADApp.so`App::AtomicPropertyChangeInterface<App::PropertyListsT<App::Material, std::vector<App::Material, std::allocator<App::Material>>, App::PropertyLists>>::AtomicPropertyChange::tryInvoke(this=0x00007fffffffb0f0) at Property.h:402:34
    frame #4: 0x00007ffff603e1a9 libFreeCADApp.so`App::PropertyListsT<App::Material, std::vector<App::Material, std::allocator<App::Material>>, App::PropertyLists>::setValues(this=0x000055555accf368, newValues=size=1) at Property.h:510:24
    frame #5: 0x00007fff9507d022 PartDesignGui.so`PartDesignGui::ViewProviderBody::copyColorsfromTip(this=0x000055555acceeb0, tip=<unavailable>) at ViewProviderBody.cpp:263:40
    frame #6: 0x00007fff9507e064 PartDesignGui.so`PartDesignGui::ViewProviderBody::updateData(this=0x000055555acceeb0, prop=0x000055555a484ee8) at ViewProviderBody.cpp:252:30
    frame #7: 0x00007ffff709b157 libFreeCADGui.so`Gui::ViewProvider::update(this=0x000055555acceeb0, prop=0x000055555a484ee8) at ViewProvider.cpp:308:15
    frame #8: 0x00007ffff70ac91b libFreeCADGui.so`Gui::ViewProviderDocumentObject::update(this=0x000055555acceeb0, prop=0x000055555a484ee8) at ViewProviderDocumentObject.cpp:383:29
    frame #9: 0x00007ffff6cceb4e libFreeCADGui.so`Gui::Document::slotChangedObject(this=0x00005555594bc390, Obj=0x000055555a484410, Prop=0x000055555a484ee8) at Document.cpp:794:33
    frame #10: 0x00007ffff6ce4c05 libFreeCADGui.so`void std::__invoke_impl<void, void (Gui::Document::*&)(App::DocumentObject const&, App::Property const&), Gui::Document*&, App::DocumentObject const&, App::Property const&>((null)=__invoke_memfun_deref @ 0x00007fffffffb490, __f=<unavailable>, __t=<unavailable>, (null)=<unavailable>, (null)=<unavailable>) at invoke.h:74:46
    frame #11: 0x00007ffff6ce4c2e libFreeCADGui.so`void std::_Bind<void (Gui::Document::* (Gui::Document*, std::_Placeholder<1>, std::_Placeholder<2>))(App::DocumentObject const&, App::Property const&)>::operator()<App::DocumentObject const&, App::Property const&, void>(App::DocumentObject const&, App::Property const&) [inlined] std::__invoke_result<void (Gui::Document::*&)(App::DocumentObject const&, App::Property const&), Gui::Document*&, App::DocumentObject const&, App::Property const&>::type std::__invoke<void (Gui::Document::*&)(App::DocumentObject const&, App::Property const&), Gui::Document*&, App::DocumentObject const&, App::Property const&>((null)=0x000055555a484ee8, (null)=0x000055555a484410, (null)=<unavailable>, __fn=<unavailable>) at invoke.h:96:40
    frame #12: 0x00007ffff6ce4c23 libFreeCADGui.so`void std::_Bind<void (Gui::Document::* (Gui::Document*, std::_Placeholder<1>, std::_Placeholder<2>))(App::DocumentObject const&, App::Property const&)>::operator()<App::DocumentObject const&, App::Property const&, void>(App::DocumentObject const&, App::Property const&) [inlined] void std::_Bind<void (Gui::Document::* (Gui::Document*, std::_Placeholder<1>, std::_Placeholder<2>))(App::DocumentObject const&, App::Property const&)>::__call<void, App::DocumentObject const&, App::Property const&, 0ul, 1ul, 2ul>((null)=<unavailable>, __args=0x00007fffffffb490, this=<unavailable>) at functional:506:24
    frame #13: 0x00007ffff6ce4c1f libFreeCADGui.so`void std::_Bind<void (Gui::Document::* (Gui::Document*, std::_Placeholder<1>, std::_Placeholder<2>))(App::DocumentObject const&, App::Property const&)>::operator()<App::DocumentObject const&, App::Property const&, void>(this=<unavailable>, (null)=0x000055555a484410, (null)=0x000055555a484ee8) at functional:591:32
    frame #14: 0x00007ffff6ce4c3d libFreeCADGui.so`boost::detail::function::void_function_obj_invoker2<std::_Bind<void (Gui::Document::* (Gui::Document*, std::_Placeholder<1>, std::_Placeholder<2>))(App::DocumentObject const&, App::Property const&)>, void, App::DocumentObject const&, App::Property const&>::invoke(function_obj_ptr=<unavailable>, a0=<unavailable>, a1=<unavailable>) at function_template.hpp:158:11
[... cut ...]
    frame #85: 0x00007ffff3229e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:107:1), argc=1, argv=0x00007fffffffdd58, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd48) at libc-start.c:392:3
    frame #86: 0x0000555555569c05 FreeCAD`_start + 37
(lldb)
```

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37665 (Git)
Build type: Debug
Branch: main
Hash: 62463fa0bd2c6962550ef67f94ca527b81b21cd9
Python 3.10.12, Qt 5.15.14, Coin 4.0.2, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

maxwxyz, 2024-06-13:
@bgbsww is this related to TNP changes?

wwmayer, 2024-06-14:
https://github.com/FreeCAD/FreeCAD/pull/14679

closed 2024-06-17 by chennes