FreeCAD/FreeCAD #14600: STEP file that imported in 20.1 shows issues in 21.1 / 21.2

Azzar2, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I use FreeCad to test gear STEP files I export from my app. Until and including 20.2, no problems. From 21.1, B_Spline_Surfaces with curvature show as flat faces.

[image]
[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * Help 1.0.3
[/code]

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

FEM

### Anything else?

[STEP Pinion-Internal_1.3.8_Microgeo1_SD_0_back_angle_wo_CID_side_BA-ForgedFillet-Solid.zip](https://github.com/user-attachments/files/15778727/STEP.Pinion-Internal_1.3.8_Microgeo1_SD_0_back_angle_wo_CID_side_BA-ForgedFillet-Solid.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-11:
can you try this in the latest dev build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Azzar2, 2024-06-11:
Good day Mr Wilfinger,

Thank you for the quick return. I downloaded and extracted the 22.Dev
update recommended (see below).

[image: image.png]

Here is what I get:

22.dev - GitHub Weekly build: seems a bit of an issue ;-()

[image: image.png]

FreeCad 20.2: all surfaces show the correct curvatures

[image: image.png]

FreeCad 21.1: all surfaces appear without curvature.

[image: image.png]

I attach my STEP file for your own tests if required.

Let me know if I can be of further assistance.

Cordialement / Best regards - Claude Gosselin
_________________________________________

*Claude Gosselin, Ph.D., P.Eng.Involute Simulation Softwares *
1139 Av. des Laurentides
Quebec, P. Quebec, Canada G1S-3C2
Email:          ***@***.***
Web:            www.HyGEARS.com
_________________________________________

This e-mail message and any attachments may contain legally privileged,
confidential or proprietary Information, or information otherwise protected
by law of Involute Simulation Softwares Inc., its affiliates, or third
parties. This notice serves as marking of its "Confidential" status as
defined in any confidentiality agreements concerning the sender and
recipient.

If you are not the intended recipient(s), or the employee or agent
responsible for delivery of this message to the intended recipient(s), you
are hereby notified that any dissemination, distribution or copying of this
e-mail message is strictly prohibited. If you have received this message in
error, please immediately notify the sender and delete this e-mail message
from your computer.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

maxwxyz, 2024-06-11:
@Kuzma30 you have an open PR for STEP import, can you have a look at this?

Kuzma30, 2024-06-11:
I try open on my test build 
[image]

OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37718 (Git)
Build type: Debug
Branch: master
Hash: 8c6c174fc11e7f9e19386552509ed4b3714a846c
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1.dev
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * sheetmetal 0.4.14
  * fasteners 0.5.22
  * Assembly3 0.12.2
  * addFC 0.2.2

maxwxyz, 2024-06-11:
@Azzar2 your images are not visible, try to comment directly on GitHub. Is the above screenshot the intended behavior?

Azzar2, 2024-06-11:
Yes - this is exactly what I was getting on 20.2.

Kuzma30, 2024-06-11:
Its build with  OCC 7.8.1

Azzar2, 2024-06-11:
I am using Win11 - seems you are on Ubuntu ... so what am I to do, apart wait for the next update ? Mind you - this is not a significant issue overall but 21.1 appears not to import the same ay 20.2 does.

maxwxyz, 2024-06-11:
@adrianinsaval what OCC version will be used for the stable release of 1.0?

adrianinsaval, 2024-06-12:
Hopefully 7.8.1 but we are currently using 7.7.2

FEA-eng, 2024-06-15:
It seems to be fine in 0.22.

Original:

[image]

Patterned (btw. it's good that we have the _Transform body_ mode now since otherwise PD patterns wouldn't work on this as there are no additive and subtractive features, just the base feature):

[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

Azzar2, 2024-06-15:
> It seems to be fine in 0.22.
>
> Original:
>
> tooth.JPG (view on web)
>
> Patterned (btw. it's good that we have the *Transform body* mode now
> since otherwise PD patterns wouldn't work on this as there are no additive
> and subtractive features, just the base feature):
>
> teeth.JPG (view on web)
>
> OS: Windows 10 build 19045
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.37730 (Git)
> Build type: Release
> Branch: main
> Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
> Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
> Locale: Polish/Poland (pl_PL)

Thank you for the follow up. When is 0.22 scheduled for release ?

Cordialement / Best regards - Claude Gosselin
_________________________________________

*Claude Gosselin, Ph.D., P.Eng.Involute Simulation Softwares *
1139 Av. des Laurentides
Quebec, P. Quebec, Canada G1S-3C2
Email:          ***@***.***
Web:            www.HyGEARS.com
_________________________________________

This e-mail message and any attachments may contain legally privileged,
confidential or proprietary Information, or information otherwise protected
by law of Involute Simulation Softwares Inc., its affiliates, or third
parties. This notice serves as marking of its "Confidential" status as
defined in any confidentiality agreements concerning the sender and
recipient.

If you are not the intended recipient(s), or the employee or agent
responsible for delivery of this message to the intended recipient(s), you
are hereby notified that any dissemination, distribution or copying of this
e-mail message is strictly prohibited. If you have received this message in
error, please immediately notify the sender and delete this e-mail message
from your computer.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

FEA-eng, 2024-06-15:
> When is 0.22 scheduled for release ?

There's no release date yet but the current plan is to release it (as 1.0) in July. I would expect some delay though so IMO August seems more likely.

Azzar2, 2024-06-15:
Thanks for this info. I will keep on using 0.20.2 till then.

maxwxyz, 2024-06-16:
I'll close this then if fixed in 0.22

closed 2024-06-16 by maxwxyz