FreeCAD/FreeCAD #14602: "Invalid shape name ?Edge6" in sketch when completely different external geometry edge edge was referenced

nazar-pc, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Here is the file: [Core.zip](https://github.com/user-attachments/files/15779522/Core.zip)

`Edge32` there was projected as external geometry from `Motherboard` sketch into `IoShield` and used there. Everything was great, I saved the file. When I reopen the file again and open/close `IoShield` sketch though I get following errors:
```
01:01:34  Invalid shape name ?Edge6
Clear constraints to external geometry
```

I didn't project `Edge6` into `IoShield` though.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

nazar-pc, 2024-06-11:
Another issue of a similar nature (I think) [Core.zip](https://github.com/user-attachments/files/15780313/Core.zip)

Everything looks great and was great before closing and reopening. Now if I open `IoShieldBase` sketch and close it, `IoShieldPanel` that references its point breaks with following:
```
03:10:10  <PropertyLinks> PropertyLinks.cpp(453): Core#Sketch006.Support missing element reference Core#Sketch005 ;g34847v1;SKT.Vertex1
03:10:10  <PropertyLinks> PropertyLinks.cpp(453): Core#Sketch006.ExternalGeometry missing element reference Core#Sketch005 ;g34847v2;SKT.Vertex2
03:10:10  Invalid shape name ?Vertex2
Clear constraints to external geometry
```

maxwxyz, 2024-06-11:
@bgbsww FYI
Is this a duplicate of https://github.com/FreeCAD/FreeCAD/issues/14563

nazar-pc, 2024-06-11:
To be clear the app doesn't crash, it just results in errors and sketch no longer being fully constrained

nazar-pc, 2024-06-11:
Happens on all sketches with `Map Mode: Translate`. Sketched that have `Map Mode: Deactivated` are not affected.

rvb-033, 2024-06-15:
I just noted the same problem, but it happens even if the Map Mode is deactivated.

In the appended file, sketch001 is constrained by references in Sketch. If one tries to edit Sketch, the references are lost. Map Mode of Sketch and Sketch001 are deactivated.

[Essai.zip](https://github.com/user-attachments/files/15847929/Essai.zip)

Regards

nazar-pc, 2024-06-16:
Okay, I don't think FreeCAD is usable for me until this is fixed. So many things fall apart all the time due to this.

FlachyJoe, 2024-06-16:
> I don't think FreeCAD is usable for me until this is fixed.

The FreeCAD version you **test** is a WIP. Last stable is 0.21.2 and FreeCAD is fully **usable** at this state as long as you take care of the TNP.

rvb-033, 2024-06-16:
Hello everybody,

As for me, I am back to 0.21.2.

How do we know that this issue of **0.22dev** is solved ?

Regards,

FlachyJoe, 2024-06-16:
> How do we know that this issue of **0.22dev** is solved ?

When this issue will be closed.

nazar-pc, 2024-06-16:
> The FreeCAD version you test is a WIP. Last stable is 0.21.2 and FreeCAD is fully usable at this state as long as you take care of the TNP.

Of course, I'm aware, makes perfect sense. TNP is something that seems to inevitably pop up in larger projects and this is why I wanted to test and report issues that I found. I intend to have a lot of dependant objects in a larger hobby open source project, hence trying to use this version of FreeCAD specifically or else I'll pull my hair out when need to make changes :slightly_smiling_face:

bgbsww, 2024-06-16:
The bug report is appreciated, and is under active work.  We just don't want anyone getting themselves in trouble because they relied on a dev version :)  Thanks again!

For the record, any name that starts with a ? is because we tried to look it up, and it wasn't there - ?Edge6 means we looked for the external reference and could not find a 6th edge in the geometry of the sketch at the time we looked.

One way to cause this is to delete geometry; and sometimes for us to fail to adjust the pointers to where the geometry went.  There may be other ways to trigger this, and we'll see if this is one of those.

FlachyJoe, 2024-06-16:
@bgbsww shouldn't we ask the user to pick the correct reference when the algo can't find it by itself?

nazar-pc, 2024-06-16:
My original issue is that reference was selected correctly and worked until I close the project and reopen it again. Somewhere in the chain of save/close/reopen doesn't quite work correctly and results in perfectly valid project suddenly becoming invalid later.

But according to Discord the cause was already found, so what is left is to wait for fix to materialize.

bgbsww, 2024-06-16:
@flachyjoe that's a neat idea that might apply in this case; although it seems like we might just be able to choose the right edge.

In the delete case, all current examples of the code just invalidate the reference.  I'm unclear on how we would trigger a dialog at the moment we are running a recompute based on closing a sketch, but there is probably a way.

rvb-033, 2024-06-18:
Trying to help : this does not happen if the two sketches are part of a body.

To be fair and fully transparent : I did not find it myself, someone else mentioned it on the French Forum.

Regards

bgbsww, 2024-06-24:
Thanks!  Because this is a save/restore issue, and because the example file in question has no ElementMaps in it when restored, I'm suspicious this is in the category of bugs solved by https://github.com/FreeCAD/FreeCAD/pull/14898 (which hasn't committed yet).  It would be useful to know if this problem is fixed.

nazar-pc, 2024-06-24:
Will test once merged (or if there is a possibility to create a test build from that branch I can test sooner).

jffmichi, 2024-06-26:
I'm not sure the issue is completely fixed with [#14898](https://github.com/FreeCAD/FreeCAD/pull/14898). I'm on revision 37914. When opening the attached file [tnp_issue.zip](https://github.com/user-attachments/files/15992504/tnp_issue.zip) and changing anything in `Sketch`, the reference edge in `Sketch001` becomes invalid. After recreating the reference, you can make changes to `Sketch` and they are reflected in `Sketch001`, save and open the file again and the issue reappears.

> OS: Arch Linux (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.37914 (Git)
> Build type: Release
> Branch: main
> Hash: 6fa8e4ccfbb7e7479521b31910d3c4ed3e6b85ea
> Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
> Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
> Installed mods: 
>   * kicadStepUpMod 11.1.7
>   * dodo 1.0.1
>   * sheetmetal 0.4.16
>   * fcgear 1.0.0
>   * fasteners 0.5.21
>   * Assembly4 0.50.12
>   * Curves 0.6.37
>   * lattice2 1.0.0

bgbsww, 2024-06-28:
Confirmed, you get:
14:38:15  Invalid shape name ?Edge4
Clear constraints to external geometry

Although if you first enter Sketch001 you'll get:
14:33:48  <PropertyLinks> PropertyLinks.cpp(433): tnp_issue#Sketch001.ExternalGeometry auto change element reference tnp_issue#Sketch ;g3337;SKT.Edge4 -> ;g8;SKT.Edge4
And then everything is fine.

That's consistent either with overstoring information in the file, or failing to run an update on the load.  On to find which...

nazar-pc, 2024-06-29:
My initial situation seems to be mostly resolved in latest dev build, there are just a few odd things. This when loading document:
```
23:57:25  <ElementMap> <string>(1): No hasherRef
23:57:25  <ElementMap> <string>(1): No hasherRef
```

And then when opening/closing sketch like `IoShieldPanel` triggers this message, though I don't understand what the actual problem is:
```
23:58:34  App::DocumentObjectGroup: Link(s) to object(s) 'Sketch' go out of the allowed scope 'Group002'. Instead, the linked object(s) reside within 'Body'.
```

Otherwise it works, here is updated/fixed document: [Core.zip](https://github.com/user-attachments/files/16042186/Core.zip)

bgbsww, 2024-06-30:
Thank-you!  Please speak up if you see something.  Because the load/save was "suboptimal" ( == "completely !@#@ up" ) we could definitely have generated some files that were broken, but then okay after saving on a good version.  It is better because of the feedback.

bgbsww, 2024-07-02:
@maxwxyz  I think this is done, no?  nazar-pc's last message indicated this is no longer replicable after the load/save fix.

nazar-pc, 2024-07-02:
I'll close this, I'm sure you're aware of other edge-cases, but the one I reported here specifically seems to be resolved. Thank you!

closed 2024-07-02 by nazar-pc