FreeCAD/FreeCAD #14603: Can't create a sketch in a group

nazar-pc, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A few simple steps on Sketcher workbench:
* create group
* select created group
* create sketch

This results in confusing error:
```
Sketch Mapping. Can't map the sketch to selected object. Broken link to support subelements.
```

Expectation is that it would create a new sketch at the bottom of selected group. I can create sketch outside of the group and move it in just fine, but it should also be possible to create a sketch in the group from the start.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

nazar-pc, 2024-06-11:
On a similar note part creation succeeds when group is selected, but is placed outside of the group, while I expected it to be inside instead.

Roy-043, 2024-06-13:
There are two separate issues here. The mapping error should be prevented, that's obvious. But the requested behavior that the sketch be created inside the selected group would be a significant change that should be discussed first.

KimK, 2024-07-15:
I never got the confusing error message. It simply created a new sketch in the middle of the group, True, it was not at the tip, I had to drag it out of the group to properly place it. But that's about normal, I never saw the error message.
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38184 (Git)
Build type: Unknown
Branch: main
Hash: 463b2d94510b49fcd6b8009f942c6ba011bd332a
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)

Jonezzzzz, 2025-01-09:
I made a PR #18971 which fixes the mapping issue that occurs when attempting to create a sketch while a group is selected.

I've also noticed that when attempting to create a sketch while a body or a part is selected also results a mapping error.

Should new issues be created for them, or are they within the scope of this issue?

furgo16, 2025-01-16:
I'm not sure if it's related to this issue, but you get the same "Can't map the sketch to selected object. Broken link to support subelements." message if you select the "Assembly" object created by the Assembly Workbench and then try to create a sketch.

maxwxyz, 2025-06-19:
@tetektoza is this maybe fixed by https://github.com/FreeCAD/FreeCAD/pull/21902 ?