FreeCAD/FreeCAD #14604: Fillet failing on curved object

mattspawn1975, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I made a post on the FreeCAD Facebook group at the following link for more details. Others were able to repeat the problem in there and one user recorded the failure in a video. I forces the user to search for more timely workarounds. Some were provided in the Facebook group.

https://www.facebook.com/share/p/TTg7DAY2zShmVMoZ/?mibextid=oFDknk

A fillet is failing at roughly the halfway point when filleting a circular extrusion. 

[image]
[image]

### Full version info

```shell
It has failed on other versions I have tried as well.

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36996 (Git)
Build type: Release
Branch: main
Hash: d92f74ef019fcedbdcdc32055ab2437c950ac23b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.64a
  * Assembly3 0.12.2
  * Assembly4 0.50.12
  * CADExchanger
  * Defeaturing 1.2.2
  * Design456 0.0.1
  * ExplodedAssembly
  * fasteners 0.5.20
  * LCInterlocking
  * QuickMeasure 2022.10.28
  * WB_Organizer 2024.1.29
```

### Subproject(s) affected?

PartDesign

### Anything else?

Here is the Facebook group mentioned above if interested.

https://www.facebook.com/share/p/TTg7DAY2zShmVMoZ/?mibextid=oFDknk

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ElementW, 2025-08-14:
https://github.com/FreeCAD/FreeCAD/labels/Status%3A%20Confirmed with `1.1.0dev.42885` (2025/08/08 15:11:51). Fails with the eternally useless `Error` and `BRep_API: Command not done` messages. Smells like https://github.com/FreeCAD/FreeCAD/labels/3rd%20party%3A%20OCC to me.

Using the following example file, I collected the failure radius of fillets compared to the radius of the extruded 1mm thick semi-ring they're on, as I had a hunch they'd be related.

[fillet.FCStd.zip](https://github.com/user-attachments/files/21770483/fillet.FCStd.zip)

<details>
<summary>Square root shenaniganery</summary>

Below is the data for each ring radius with the maximum fillet radius that still works.

```csv
2000,63.25
1500,54.78
1000,44.73
500,31.63
250,22.38
100,14.17
```

Had a feeling this was a square root, and sure enough, the trend line is a power of `0.5`.

[image]

</details>

Also fiddled with the ring's thickness. Conclusion: yes, it has an influence on the max allowable radius. How much you ask? Well, notice something about how the max possible fillet radius looks when projected?

[image]

In fact this also applies to chamfers, unsurprisingly.
This seems to be an edge case making fillets/chamfers fail if edges created don't line up to some criteria I haven't completely figured out yet, but I presume it's related to the end edge that's supposed to be created.

This can be reduced down to a simple example:

[image]

Trying to fillet/chamfer the leftmost edge with more than the blue dimension, here 2 mm, fails as most likely OCCT or FC's use of it attempts to create an edge outside of the surfaces being chamfered, here shown with 3 mm.

But if you bring it back into validity, then the chamfer works:

[image]

[image]

It's very messed up and is probably the source of *many* other failures.

Tried finding duplicates of this but I see crashes and wrong results instead of failures like here.

@maxwxyz Could you reopen this one please?

maxwxyz, 2025-08-14:
@pieterhijma OCC related

ElementW, 2026-08-25:
Still relevant as of faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c and OCC 8.1.0.dev1-3d097a0328