FreeCAD/FreeCAD #14609: Assembly: Deleting an exploded view does not delete the move objects within.

PaddleStroke, 2024-06-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Deleting an exploded view does not delete the move objects within.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-16:
@PaddleStroke reopening this as it happens again in latest main https://github.com/FreeCAD/FreeCAD/issues/17902

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39176 (Git) Conda
Build type: Release
Branch: main
Hash: 45d3864942c6928cec7dc18f95b1f222485756e5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.27.14
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.3
  * Silk 0.1.5

```

furgo16, 2024-11-16:
I feel #17902 is not a duplicate. It is about the exploded views group. Whereas the reopened issue was about the individual views. As I say on #17902, deleting individual views works, so it does not appear to be a regression.

closed 2024-11-16 by maxwxyz