FreeCAD/FreeCAD #14613: New CAM Simulator, ToolBit Shapes not displayed properly

jkvfc, 2024-06-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The picture below shows the tool paths for tapping and slitting. The toolbit for tapping is self-generated and that for slitting is FreeCAD's default slitting saw.

[image]

With the **old simulator** the toolbit shape for the tapping tool is **ok**.

[image]

With the **new simulator** the shape is **wrong**.

[image]

With the **old simulator** the toolbit shape for the slitting saw is **ok**. But the **reference point** is **wrong**. It should be on the underside of the saw blade.

[image]

With the **new simulator** the output for the slitting operation fails. The error message below is displayed. 
 
[image]

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37649 (Git)
Build type: Unknown
Branch: main
Hash: 681e8c9d2f95d4341e48cec629267182b2f7e586
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * MyScripts
```

### Subproject(s) affected?

CAM/Path

### Anything else?

[TestToolsTappingSlitting.zip](https://github.com/user-attachments/files/15787216/TestToolsTappingSlitting.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-11:
@shaise FYI

shaise, 2024-06-12:
Thank you.
Taking a look.

jkvfc, 2024-06-16:
@shaise,
fix successfully tested, thank you very much for the quick solution.

closed 2024-06-16 by jkvfc