FreeCAD/FreeCAD #14616: Union of lofted solid bodies is creating surface errors

KushalBosu, 2024-06-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[Freecad Union Issue.docx](https://github.com/user-attachments/files/15791901/Freecad.Union.Issue.docx)
A lofted solid body is created from two non-planar and non-parallel circles in 3D.. Similarly, another lofted solid body is created using 2 more circles. When a union operation is done using these two solid bodies, a surface error is created. But when these two solid bodies are combined in the Ansys spaceclaim software, there is no error.
I think there is a problem with the tolerance of the combined surface that results from union. This tolerance should be provided in the GUI.

### Full version info

```shell
FreeCAD 0.20.2
Windows
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-11:
Please add an example file, ideally also screenshots in the issue and full about info.
You are using an old version, could you try this in the latest dev build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

maxwxyz, 2024-06-15:
@KushalBosu please add more info or we'll close this issue

luzpaz, 2024-07-13:
@KushalBosu please respond

KushalBosu, 2024-07-13:
close the issue

closed 2024-07-13 by maxwxyz