FreeCAD/FreeCAD #14618: [Sketcher] Regression using Edit Mode > Rubber Band Selection major drop in performance

Syres916, 2024-06-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a benchmarking macro that generates a large quantity of geometry and selecting a quarter of it took _3.52 seconds_ using build 0.22.0dev.36874. It now takes **_9 minutes 27.96 seconds_** 

[image]

While this maybe an extreme case, it was while carrying out some end-to-end testing with significant geometry that this situation became apparent. I've eliminated the recent PR https://github.com/FreeCAD/FreeCAD/pull/14462

### Full version info

```shell
For my bisecting info:

Good/fast response build:

OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * sheetmetal 0.4.1

Bad/Slow response build
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37582 (Git)
Build type: Release
Branch: main
Hash: 6fb7c51c1827d4e3989e2d562292cae8dee6d93f
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * sheetmetal 0.4.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-06-12:
This is not a TNP or Sketcher Selection mechanism issue, the root cause is QuickMeasure https://github.com/FreeCAD/FreeCAD/commit/e5780a5e3d36b07b7e16abdf9d7611ca48b38421 being active in Sketcher Edit Mode

PaddleStroke, 2024-06-12:
One solution could be to set a maximum to the number of element for quickmeasure. Because I guess this could happen for large selections outside of sketcher edit mode as well.
I will check it. @maxwxyz you can assign me

PaddleStroke, 2024-06-12:
The first thing to note is that selecting 200 components is doing  200 times Gui.Selection.addSelection.

> >>> # Gui.Selection.addSelection('Unnamed','Body','Sketch.Vertex37',0,0,0,False)
> >>> # Gui.Selection.addSelection('Unnamed','Body','Sketch.Vertex38',0,0,0,False)
> >>> # Gui.Selection.addSelection('Unnamed','Body','Sketch.Vertex39',0,0,0,False)

Just this is going to make things not instantaneous. But first let's get back to few seconds rather than 10 minutes

PaddleStroke, 2024-06-12:
@wwmayer what do you think is the best solution here? Ideas that I have so far : 

A - Limit the QuickMeasure to 5 or 10 selections. `QuickMeasure::canMeasureSelection` can return false if there's more than 10 selections. Not ideal and not the faster. But self contained.

B - From `ViewProviderSketch::doBoxSelection` we could temporarily disable QuickMeasure at the beginning and do a QuickMeasure::tryMeasureSelection at the end. This way QuickMeasure is executed only once no matter how many selection.
- Problem : How VPSketch can access the QuickMeasure object?

C -  `ViewProviderSketch::doBoxSelection` change the way things are added to selection. Is it possible to add all the subnames in one single Gui.Selection.addSelection?
- Problem : all the selection observer in sketcher needs to be updated. Notably the TaskSketcherElement one.
- This is probably the best way as it would solve the slow down in all the selection observers.

Other ideas ?

PaddleStroke, 2024-06-12:
D - in QuickMeasure , add a timer of 200ms before executing `tryMeasureSelection` with some boolean such that it triggers only once.

wwmayer, 2024-06-12:
This doesn't only happen in sketcher edit mode but also in normal viewing mode when using the rubber band selection from the Edit menu.

> The first thing to note is that selecting 200 components is doing 200 times Gui.Selection.addSelection. So saying that this is not a Sketcher Selection mechanism issue is wrong imo. It is clearly exactly that.

It's clearly not a selection mechanism issue but the quick measure slows it down. If quick measure is disabled in the code the selection more or less is done immediately. So, calling addSelection() for every sub-shape doesn't cause the delay.

Here some test code to create a compound of 1000 line segments.
```
shapes = []
for i in range(1000):
    shapes.append(Part.makeLine(App.Vector(i, 0, 0), App.Vector(i+1, 0, 0)))

comp = Part.Compound(shapes)
shape = Part.show(comp)
```

Now when I do the rubber band selection then the executed Python code will be of this form
```
Gui.Selection.addSelection('Unnamed','Shape','Edge1')
Gui.Selection.addSelection('Unnamed','Shape','Edge2')
Gui.Selection.addSelection('Unnamed','Shape','Edge3')
Gui.Selection.addSelection('Unnamed','Shape','Edge4')
Gui.Selection.addSelection('Unnamed','Shape','Edge5')
Gui.Selection.addSelection('Unnamed','Shape','Edge6')
...
```
You have to make sure to not select the entire object as otherwise it will execute this code line
`Gui.Selection.addSelection('Unnamed','Shape')` instead.

With Python it's possible to select every sub-shape in one go.
```
edges = ["Edge{}".format(i+1) for i in range(1000)]
Gui.Selection.addSelection(App.ActiveDocument.Shape, edges)
```
When doing so and quick measure is active it also takes very long time. So again, the culprit is clearly the quick measure.

> Limit the QuickMeasure to 5 or 10 selections.

And add an option to switch it completely off.

> From ViewProviderSketch::doBoxSelection we could temporarily disable QuickMeasure

It's not only related to sketch edit mode but a general issue. Also 

> ViewProviderSketch::doBoxSelection change the way things are added to selection. Is it possible to add all the subnames in one single Gui.Selection.addSelection?

As shown above this won't help.

> in QuickMeasure , add a timer of 200ms before executing tryMeasureSelection with some boolean such that it triggers only once.

This doesn't help either. The problem simply is that getting the required information for so many shapes is very slow. It doesn't matter when it's done but that it's done. The only solution is to set an upper limit of shapes that should be processed by quick measure.

PaddleStroke, 2024-06-12:
> It's clearly not a selection mechanism issue but the quick measure slows it down. If quick measure is disabled in the code the selection more or less is done immediately. So, calling addSelection() for every sub-shape doesn't cause the delay.

I disagree. Selecting 400 points in sketcher takes several seconds even without QuickMeasure. It is not instantaneous and it is laggy.
The way the selection mechanism is adding and signaling one by one the addition seems problematic. There should be only one signal signaling that a vector of things have been added to the selection.

```
edges = ["Edge{}".format(i+1) for i in range(1000)]
Gui.Selection.addSelection(App.ActiveDocument.Shape, edges)
```
Doing this is the same as doing 
```
for edge in edges:
     Gui.Selection.addSelection(App.ActiveDocument.Shape, edge)
```
addSelections() is still sending one signal for each selection added. So it's the same it's sending 200 signals.

> This doesn't help either. The problem simply is that getting the required information for so many shapes is very slow. It doesn't matter when it's done but that it's done. The only solution is to set an upper limit of shapes that should be processed by quick measure.

No this is not the problem. Getting the information for 200 edges is actually not so long. The problem is that the code is running for every addSelection signal. So it's exponentially slower. It's like this : 

addSelection edge1
QuickMeasure clear references
- add edge1 to ref
- print the message

addSelection edge2
QuickMeasure clear references
- add edge1 to ref
- add edge2 to ref
- print the message

addSelection edge3
QuickMeasure clear references
- add edge1 to ref
- add edge2 to ref
- add edge3 to ref
- print the message

and so on until edgeN.

So in total it's doing N * (N +1) / 2 calculation of length. And printing the message N times.

If it was triggering only once, then it would need to do only N calculation of length and printing 1 time.
Which you can verify if you first select 200 edges it takes a long time. Then if you add one single edge to the selection it's instantaneous.

wwmayer, 2024-06-12:
> So it's exponentially slower.

Well, it's only polynomial but that's still too inefficient. The complexity of the algorithm is O(n**2) not O(2**n).

PaddleStroke, 2024-06-12:
Indeed 2:30 is still not good.
So what about a timer of 100ms to make sure QuickMeasure triggers only 1 time + a limit of 10 selection (selection + subnames count)

wwmayer, 2024-06-12:
I wonder whether we should extend the class SelectionChanges to allow to pass all sub-elements in one go. Maybe not now but after the release of 1.0.

> So what about a timer of 100ms to make sure QuickMeasure triggers only 1 time + a limit of 10 selection (selection + subnames count)

This sounds like a good compromise.

closed 2024-06-17 by chennes