FreeCAD/FreeCAD #14621: add a popup on first start of freecad to teach the user how to use it

Vincenzoferrara, 2024-06-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I don't know if I should have asked it here or on the forum.

in most cad software, at the first start a pop-up appears that shows you how to use the software and I think freecad should have it too.

more or less basic things, like how to personalize the graphics with themes, and change the colors.
that using a spreadsheet in sketch you can create variables.
how to change the navigation style of the 3d model.
explain that to print you have to use TechDraw.

making the user understand the infinite customization, and all the most beautiful features, can make the user choose to try to learn it

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git) AppImage
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT) [ OS: Italian/Italy (it_IT) ]
Installed mods: 
  * AutoOpenSpreadsheet
  * ProDarkThemePreferencePack 1.0.0
  * CommandPanel
  * IconThemes
  * fasteners.backup1715436808.37841 0.5.16 (Disabled)
  * FeedsAndSpeeds 0.5.0
  * ExplodedAssembly
  * OpenTheme 2024.5.16
  * Freecad-Built-in-themes-beta 1.2.2
  * fasteners 0.5.21
  * TabBar
  * Assembly4 0.50.12
  * CubeMenu
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-12:
The simplest and least invasive way to do that could be to add some short tips to the splash screen (with an option to disable them).

maxwxyz, 2024-06-13:
related https://github.com/FreeCAD/FreeCAD/issues/11735

luzpaz, 2024-07-13:
Should this be merged into #11735

luzpaz, 2025-04-29:
Isn't there an ongoing funded FPA grant for something like this ?