FreeCAD/FreeCAD #14623: Integreted Assembly WB doesn't allow a Link with multiple Element count

MiqCG, 2024-06-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Integrated Assembly WB already allows the link of mirrored parts with the Scale = -1 feature.  Very useful.

On the other hand looks like it is not possible to multiply a linked part using the link feature "Element count" (number different of zero). The Joint Coordinate System (JCS) don't attach to the multiplied part. Same behaviour if you multiply the linked part before or after dragging into the Assembly container.

I think it would be nice to have this feature because working with links is fast and powerful.

On the other hand, is pretty easy to have multiple copies of a component with the Insert component tool of the Integrated assembly WB. 

Looking forward to hear opinions about this.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git) AppImage
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * fasteners 0.5.22
```

### Subproject(s) affected?

Assembly

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/18265355/a78ed4f4-fa14-4a34-a3de-baded76e76d0

[Github_El_Count.zip](https://github.com/user-attachments/files/15795423/Github_El_Count.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-12:
Support for link groups need to be implemented then.

Anthony-Gaudino, 2024-08-06:
Just giving my +1 to this feature.
Assembly3 handles links with multiple copies, this is very useful. I have some models with 40 or more link copies.

closed 2024-09-02 by yorikvanhavre