FreeCAD/FreeCAD #14628: Chamfer can produce strange artifacts

Nashenas88, 2024-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Adding a chamfer to these features causes the addition of a large circular plane artifact. It becomes part of the model, e.g. I can see it if I export the model to stl and load it in another program.

Here I chamfer the edges on one of the tabs of the model.
[image]

Here's what it looks like when I add a chamfer on the edges of all the tabs.
[image]

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fcgear 1.0.0
  * Dracula 0.0.2
  * CurvedShapes 1.0.4
  * Assembly4 0.12.4
  * Curves 0.6.4
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

Zipped project file that can reproduce the issue.
[moon-base-bug.zip](https://github.com/user-attachments/files/15798096/moon-base-bug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-06-13:
Chamfers are finicky at the best of times in OCC/FreeCAD, and this looks like a classic CAD kernel issue, but I think the issue you may be having here stems as much from your design.

There is a gap underneath the lower edge of the tab:

[image]

The small height of this presumably unneeded face is why you can't get a chamfer to work beyond about 0.25mm. Experts here can explain better but my understanding of it is that chamfers can't "consume" existing edges.

If you redesign so that chamfer runs up all the way to the edge of the loft, you might be able to get larger chamfers without error.

depthoffocus, 2024-06-13:
Update. If you:

1) redesign so that chamfer runs up all the way to the edge of the loft as mentioned,

2) shift the pad's sketch infinitesimally upwards so it is not coplanar with the bottom of the ring (Z: -0.001mm or whatever)

You should be able to get larger chamfers without avoiding OCC chamfer issues.

[image]

maxwxyz, 2024-06-15:
Thanks for the help, I'll close this issue.

closed 2024-06-15 by maxwxyz

depthoffocus, 2024-06-15:
> Thanks for the help, I'll close this issue.

I mean, it is still a bug! :-) But I think it's probably one that merges with an existing report.

Nashenas88, 2024-06-22:
@maxwxyz I agree with depthoffocus that this is something a regular user should not encounter. I appreciate his help in sharing a temporary workaround. Shouldn't this be re-opened?