FreeCAD/FreeCAD #14629: Mirror fails in some situations

chopsywa, 2024-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a body that will not mirror. I have tested some other simple items and they will mirror fine, but in some circumstances it fails. I have attached a file with a body that will not mirror. The pad mirrors fine, but after I add the pocket it will no longer mirror. I hope this is enough information.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * Assembly3 0.12.2
  * Assembly4 0.50.13
  * Curves.backup1706353084.2536438 0.6.13
  * fasteners 0.5.21
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

PartDesign

### Anything else?

Don't seems to get any errors in Report view. End up with a mirrored item in workflow, but it is just the original half.
[TestCoverClipPipe.zip](https://github.com/user-attachments/files/15799159/TestCoverClipPipe.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-15:
@chopsywa please add more infos how to reproduce, it perfectly mirrors:
[image]

chopsywa, 2024-06-16:
I am now on build 37709 and I get the same result as before. Attached is a quick video of me trying to do the mirror.

https://github.com/FreeCAD/FreeCAD/assets/109642777/ac57c0f8-d08c-4ee0-b628-9e120af58f69

chopsywa, 2024-06-17:
I just had another go and it did work. It was my mistake. I left the setting to Transform Tool shapes. This setting is OK if I set the tip to the Pad above, but I have to change it to Transform Body on the Pocket. I don't really understand this setting, but it is my mistake. Sorry for the hassle.