FreeCAD/FreeCAD #14633: Harmonize Add property dialogs

furgo16, 2024-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Properties can be added to a `VarSet` in at least two ways:

1. By creating a VarSet via the `Std_VarSet` command in the Structure toolbar, then subsequently double-clicking on the created VarSet on the Tree view every time a property needs to be added
2. By using the Property editor: on the VarSet created, right-clicking on an empty area of its Property view and choosing the <kbd>Add property</kbd> context menu entry.

Using either method, the user is presented with similar but different "Add property dialogs":
- Their fields are ordered differenty
- Some of the fields are different. Sometimes present in one and not in the other (e.g. Value, Tooltip, Documentation)
- Some of the fields use different Qt widgets (e.g. Type)

| Dialog | Notes |
|---|---|
|[image] | **VarSet dialog (1)**<br>- Allows successive input of properties<br>- Allows input of value upon creation<br>- Does not allow saving property until name given<br>- Has tooltip field (and the tooltip is not shown anywhere?)  |
|[image] | **Property editor dialog (2)**<br>- Allows prefixing group name<br>- Has documentation field, which is displayed as a tooltip in the Property view |

Unless there is a specific reason for them to differ, from a usability point of view, it would be optimal that the two dialogs were the same. I guess from a code maintenance point of view, the code could be reused to avoid duplication as well.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37710 (Git) Snap 1048
Build type: Release
Branch: main
Hash: a6937d8d2677594f548af218cc92f797d9a475d9
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * Curves 0.6.37
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.3
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-13:
@pieterhijma FYI

pieterhijma, 2024-06-13:
Thanks for reporting. I'm totally aware of this. The difference is that the property view dialog is generic, so it works for properties of all types of document objects, not only VarSets.

With VarSets I'm promoting parameterized design by means of properties (instead of spreadsheets), so I wanted to have a way to add values immediately, which is not straightforward because you first need to know the type of the property. In that sense, having the ability to create a property and add a value in one step/dialog is new in FreeCAD and only possible with this specific dialog for VarSets (and the Property Manager in https://github.com/FreeCAD/FreeCAD/pull/13112).

It would be good to discuss whether we should make the dialogs the same, indeed.

krushia, 2024-06-14:
They should be the same in my opinion. I'm here because I've noticed the VarSets dialog is missing the "Allows prefixing group name" feature. The single line for tooltip also makes it difficult (impossible?) to enter multi-line text.

The Expression dialogs are also duplicated. Sometimes I get the VarSets version, sometimes the "original" version. It seems to depend on the property type.

krushia, 2024-06-15:
While I'm here, there is one tiny thing that DynamicData's version has. There is a "typing hint" in the empty Value box for most types.
[image]

furgo16, 2024-06-15:
@pieterhijma thank you for the reply, and for introducing VarSets!

> Thanks for reporting. I'm totally aware of this. The difference is that the property view dialog is generic, so it works for properties of all types of document objects, not only VarSets.

My assumption is that this difference is not relevant from a usability standpoint. I.e. the user does not see this distinction, but sees two dialogs for no apparent reason (even if there is one).

Setting the "Add Another" one aside, each dialog has 5 fields, 3 of which are the same. Would merging the two dialogs be an option?

> With VarSets I'm promoting parameterized design by means of properties (instead of spreadsheets), so I wanted to have a way to add values immediately, which is not straightforward because you first [...]

Would it not be possible to do the same in the original "Add property" dialog by reordering the fields and adding the code to dynamically create the Value field according to the property type?

> It would be good to discuss whether we should make the dialogs the same, indeed.

Indeed, it'd be good to hear the opinion of the DWG as well. Even after understanding why there are two different dialogs, I'm still of the opinion that having a consistent workflow with the same dialog would be more beneficial to the user.

Regarding the tooltip field question: where is the tooltip supposed to be shown? After setting a tooltip using the VarSet Add Property dialog, I have not found out where to see it (it is not shown when hovering over the property in the Property Manager, as far as I can tell).

Thanks!

krushia, 2024-06-15:
> Regarding the tooltip field question: where is the tooltip supposed to be shown? After setting a tooltip using the VarSet Add Property dialog, I have not found out where to see it (it is not shown when hovering over the property in the Property Manager, as far as I can tell).

They should be shown when hovering mouse over the property in the Tree view. AFAIK they can't be accessed with expressions, but can be printed in the console like `obj.getDocumentationOfProperty('Label')`. You are correct in that it appears to be broken when set using the VarSets dialog.

pieterhijma, 2024-06-16:
> They should be the same in my opinion. I'm here because I've noticed the VarSets dialog is missing the "Allows prefixing group name" feature. The single line for tooltip also makes it difficult (impossible?) to enter multi-line text.

For VarSets, I find the prefixing group name not that necessary because you can group by creating new VarSets. I personally find the prefixing confusing and would rather see it gone, but it appears that you (and others) make use of it?
 
> The Expression dialogs are also duplicated. Sometimes I get the VarSets version, sometimes the "original" version. It seems to depend on the property type.

Well-spotted! It depends on whether there is a VarSet in any document to add properties to. The reason is that I don't want to force anyone to use VarSets, so if a user has a workflow without VarSets, the expression dialog is the same as before.

pieterhijma, 2024-06-16:
> While I'm here, there is one tiny thing that DynamicData's version has. There is a "typing hint" in the empty Value box for most types. ![Screenshot_20240614_195825](https://private-user-images.githubusercontent.com/2147138/339934233-b067beeb-9d0e-4a56-88d6-2588d5db0ee2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg1MzczMTQsIm5iZiI6MTcxODUzNzAxNCwicGF0aCI6Ii8yMTQ3MTM4LzMzOTkzNDIzMy1iMDY3YmVlYi05ZDBlLTRhNTYtODhkNi0yNTg4ZDVkYjBlZTIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MTZUMTEyMzM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzJhODljZjM1ZjRhOWNiZmI4ODdmZjYyNDRiZDAyMTFmYjRmNTVmM2IxN2JjZTg3NjEzOGFhYTNiMjljMTFmMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.q3J_HFjo2587VyIVjMFMeDyGtv6m6qqAvEYmzbo9ZvA)

Yes, dynamic data uses other mechanisms to set a value, not using the built-in "editors" that FreeCAD already has in core.

pieterhijma, 2024-06-16:
> @pieterhijma thank you for the reply, and for introducing VarSets!
> 
> > Thanks for reporting. I'm totally aware of this. The difference is that the property view dialog is generic, so it works for properties of all types of document objects, not only VarSets.
> 
> My assumption is that this distinction is not relevant from a usability standpoint. I.e. it's an implementation detail: the user is oblivious to the difference, yet they see (and are surprised by) two different dialogs for no apparent reason (even if there is one).
> 
> Setting the "Add Another" one aside, each dialog has 5 fields, 3 of which are the same. Would merging the two dialogs be an option?

Totally. The fact that there are two versions is an artifact of open source development. I introduced a feature that is independent of other FreeCAD logic. You are all now participating in this process and raising the question why there are two different ways of doing the same thing. So, first of all, thank you very much. It really helps making things better.
 
> Would it not be possible to do the same in the original "Add property" dialog by reordering the fields and adding the code to dynamically create the Value field according to the property type?

Yes, that would certainly be possible.
 
> Indeed, it'd be good to hear the opinion of the DWG as well. Even after understanding why there are two different dialogs, I'm still of the opinion that having a consistent workflow with the same [...]

Ok, let's ping the @FreeCAD/design-working-group right now.

> Regarding the tooltip field question: where is the tooltip supposed to be shown? After setting a tooltip using the VarSet Add Property dialog, I have not found out where to see it (it is not shown when hovering over the property in the Property Manager, as far as I can tell).

You're right, it doesn't show. I'll make a note of it and file a different issue and PR to fix it once this discussion has settled.

pieterhijma, 2024-06-16:
From the above, let me try to document what the consensus seems to be right now:

1) The "Add property to VarSet" and "Add property" dialogs should be the same (effectively there is only one "Add property" dialog).

2) The resulting dialog should allow to enter a value like the "Add property to VarSet" dialog.

3)  There should be a Documentation field like "Add property" that allows multi-line input (and results in a tooltip).

4) There should be a checkbox to add other values as well similar to "Add property to VarSet".

5) There should be a checkbox to add group prefixes similar to "Add property"?

What about other differences?

6) The Group field should auto complete similar to "Add property to VarSet"

7) The Type field should auto-complete similar to "Add property to VarSet"

By the way, my consideration for a single line Tooltip instead of a multi-line was that it allows you to press Enter to add various properties fast.

kadet1090, 2024-06-16:
> 4\. There should be a checkbox to add other values as well similar to "Add property to VarSet".

What about two buttons. `Add` and `Add and create next`? This seems like better UX solution.

> It may be possible to allow a multi-line input box and still allow one to press Enter to add properties fast.

Maybe this should be input that automatically extends? Something like github comment. Then we can do standard behavior of enter accepts and alt+enter creates new line.

pieterhijma, 2024-06-19:
Thanks for the input!

> > 4. There should be a checkbox to add other values as well similar to "Add property to VarSet".
> 
> What about two buttons. `Add` and `Add and create next`? This seems like better UX solution.

Yes, makes sense.
 
> > It may be possible to allow a multi-line input box and still allow one to press Enter to add properties fast.
> 
> Maybe this should be input that automatically extends? Something like github comment. Then we can do standard behavior of enter accepts and alt+enter creates new line.

Ok, I can take a look.

furgo16, 2024-07-06:
The summary (with `kadet1090`'s addition) sounds great, thanks.

> For VarSets, I find the prefixing group name not that necessary because you can group by creating new VarSets. I personally find the prefixing confusing and would rather see it gone, but it appears that you (and others) make use of it?

I do not personally use the prefix, so I don't have a strong opinion on that one. I simply listed the differences to highlight the inconsistencies. To me, the main point is that the dialogs should be the same (and if technically possible the same code) to avoid user confusion. There is at least one example in the forum where someone was baffled at settings present in one dialog and not the other (group prefix), yet not realizing there are two dialogs: https://forum.freecad.org/viewtopic.php?style=5&p=769527#p769527

pieterhijma, 2025-01-13:
I'm about to work on this issue and I'm trying to resolve the discussion.

I think we have almost consensus on this issue.  I believe what we currently want is:

1) The "Add property to VarSet" and "Add property" dialogs should be the same (effectively there is only one "Add property" dialog).

2) The resulting dialog should allow to enter a value like the "Add property to VarSet" dialog.

3) There should be a Documentation field like "Add property" that allows
   multi-line input (and results in a tooltip).  Preferably an input that
   automatically extends, similar to a GitHub comment.  Enter accepts and moves
   to the new files, Alt+Enter adds a new line.

4) There should be an "Add" button that closes the dialog and an "Add and create next" button that resets the dialog for new input.

5) The Type field should auto-complete similar to "Add property to VarSet".

6) The Group field should auto complete similar to "Add property to VarSet".

We don't have consensus on the group prefix checkbox, (see #17919) and I will create a forum topic for that to
understand from the community what the preference is.

furgo16, 2025-04-28:
@pieterhijma it would seem there is consensus on 90% of what needs to be done, and there is a forum discussion for the rest.

I would propose:

1. Implementing the consensuated points
1. Removing the group prefix checkbox.

If there is wide demand for it at some point, I guess it can be brought back. This is a similar approach as the one followed when removing the Custom folders property from the 1.0 Start page (first removed, then [brought back](https://github.com/FreeCAD/FreeCAD/pull/19473)).

I feel this would be a much more satisfactory outcome than blocking the result on the group prefix. In my opinion, the group prefix option is the smallest of the issues here.

Most importantly, there is already a precedent of removing it. It's been unavailable in VarSets since their inception during the 1.0 development cycle, before November '24. So far there hasn't been an outcry for having it on the VarSet dialog (as far as I can tell, I could also be wrong).

This might also be a topic where the @FreeCAD/design-working-group might want to weigh in on.

Thoughts?

pieterhijma, 2025-05-01:
This is not blocking on the group prefix in my view. I think the consensus is to get rid of the group prefix and I agree that noone complained that the VarSet dialog doesn't have it. I'm currently working on improving the VarSet dialog and then I will replace the Property View dialog with the VarSet one, leading to having only one dialog.

luzpaz, 2025-05-05:
Note: please be aware of #19820

PR #23426 (2025-09-13): Gui: Reconcile Add Property Dialogs
This PR removes the old "Add Property" dialog in favor of the "Add Property" dialog of VarSets.

The VarSet dialog is now generic and works for any document object or document. In documents, you don't have to select a property any longer to add a new property.

Please note that other than the old "Add Property" dialog, this dialog does not support adding properties to multiple document objects. Since this was an obscure feature (see https://github.com/FreeCAD/FreeCAD/issues/23424) and allowing adding values directly makes the logic more complicated, I believ [... cut]