FreeCAD/FreeCAD #14634: [Problem] Escape (Esc) keystroke doesn't exit Sketcher if "nothing is touched"

MiqCG, 2024-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Having marked the option "Esc can leave sketch edit mode" in Edit > Preferences > Sketcher > General > General menu,
when entering Sketcher double clicking an existing sketch, if you hit Escape key immediately after, the Sketcher doesn't close, no matter how many times you hit it. 
It do exit if any function or dragging is done.

Is this the expected behaviour or is it a bug?

I have search in Github [is:issue esc sketcher] with no relevant results.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git) AppImage
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * BillOfMaterials 0.0.8.5
  * fasteners 0.5.22
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/18265355/39afb975-5156-49bc-9487-7d42f35803a9

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-13:
Hmm this bug is indeed annoying.
I think that after double clicking the sketch to enter edit mode, the 3dView does not have the focus. It's the treeView that has the focus.
And so pressing ESC is not caught by the VPSketch and so does not exit.

I need to check if we can give the focus to the 3d view at the end of the double click

PaddleStroke, 2024-06-13:
So yes there are 2 bugs reported here : 
- Start edit sketch, press ESC : does not exit edit mode
- Start a sketcher tool, press ESC : that terminates the tool. Press ESC again : does not exit edit mode

Both are a focus problem. The MDI does not have focus as user clicked somewhere outside of it. So VPSketch does not receive the ESC key signal. So it does not exit.

MiqCG, 2024-06-13:
> So yes there are 2 bugs reported here :

@PaddleStroke, do you want me to segregate this issues in two?

PaddleStroke, 2024-06-13:
No let it like this.

closed 2024-06-13 by chennes