FreeCAD/FreeCAD #14636: Measurement: Incorrect COM icon

FEA-eng, 2024-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For me, all the icons introduced with #14011 are fine, apart from COM which is just the icon of the tool itself:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-12:
@hlorus

luzpaz, 2024-07-12:
@hlorus is this on your radar ?

hlorus, 2024-07-12:
Yes it is, but it requires the viewprovider to be accessible from python. Therefore it's not a one-line fix as it might seem.

luzpaz, 2024-09-30:
Will this get a fix before 1.0 ?

hlorus, 2024-10-01:
IMO this is not the highest prio so i'm afraid it won't get fixed by me before the 1.0 release...

luzpaz, 2024-10-15:
@FEA-eng do you have a proposal of what the icon should be ?

hlorus, 2024-10-15:
There's already an icon in the codebase: https://github.com/FreeCAD/FreeCAD/blob/main/src%2FMod%2FMeasure%2FGui%2FResources%2Ficons%2FMeasurement-CenterOfMass.svg

luzpaz, 2025-04-02:
@hlorus do you have a moment to fix this ?

FEA-eng, 2025-07-05:
Still relevant although the Center of Mass mode is currently broken: #22287

closed 2025-09-03 by chennes