FreeCAD/FreeCAD #14640: Additive Loft operation causes Segmentation fault

MatejSpindler, 2024-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Additive Loft operation between a sketch and a datum line crashes FreeCAD. I know that this operation doesn't make sense, but at least it shouldn't crash FreeCAD.

Log from the crash:
```
FreeCAD 0.22.0, Libs: 0.22.0devR37707 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3cae0) [0x7fb3fe050ae0]
#1  0x7fb3e4c7c02c in BRep_Tool::Parameter(TopoDS_Vertex const&, TopoDS_Edge const&, double&) from /usr/lib/libTKBRep.so.7.7+0x1cc
#2  0x7fb3e4c7c593 in BRep_Tool::Parameter(TopoDS_Vertex const&, TopoDS_Edge const&) from /usr/lib/libTKBRep.so.7.7+0x23
#3  0x7fb3a8ffdaeb in BRepFill_CompatibleWires::Perform(bool) from /usr/lib/libTKBool.so.7.7+0x5ab
#4  0x7fb3aa31435a in BRepOffsetAPI_ThruSections::Build(Message_ProgressRange const&) from /usr/lib/libTKOffset.so.7.7+0x1ba
#5  0x7fb3ab3bf839 in Part::TopoShape::makeElementLoft(std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, Part::IsSolid, Part::IsRuled, Part::IsClosed, int, char const*) from /usr/lib/freecad/lib/Part.so+0x1d9
#6  0x7fb39307b21b in PartDesign::Loft::execute() from /usr/lib/freecad/lib/_PartDesign.so+0x92b
#7  0x7fb400cf8142 in App::DocumentObject::recompute() from /usr/lib/freecad/lib/libFreeCADApp.so+0xc2
#8  0x7fb400c7e364 in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x334
#9  0x7fb400c82d77 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /usr/lib/freecad/lib/libFreeCADApp.so+0x647
#10  0x7fb400d6c4b6 in App::DocumentPy::recompute(_object*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x2f6
#11  0x7fb400d6c64c in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x2c
#12  /usr/lib/libpython3.12.so.1.0(+0x1a52ed) [0x7fb3fffa52ed]
#13  /usr/lib/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x2fb) [0x7fb3fff8550b]
#14  /usr/lib/libpython3.12.so.1.0(+0x8bdfa) [0x7fb3ffe8bdfa]
#15  /usr/lib/libpython3.12.so.1.0(PyEval_EvalCode+0xb7) [0x7fb40003d767]
#16  /usr/lib/libpython3.12.so.1.0(+0x2608b7) [0x7fb4000608b7]
#17  /usr/lib/libpython3.12.so.1.0(+0x25b9dc) [0x7fb40005b9dc]
#18  /usr/lib/libpython3.12.so.1.0(PyRun_StringFlags+0x6d) [0x7fb40004dedd]
#19  0x7fb40075f83e in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib/freecad/lib/libFreeCADBase.so+0x6e
#20  0x7fb4018ffcc7 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x97
#21  0x7fb4018ffe63 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/lib/freecad/lib/libFreeCADGui.so+0xe3
#22  0x7fb401900007 in Gui::Command::updateActive() from /usr/lib/freecad/lib/libFreeCADGui.so+0x47
#23  /usr/lib/freecad/lib/PartDesignGui.so(+0x1387bc) [0x7fb390b387bc]
#24  /usr/lib/freecad/lib/PartDesignGui.so(+0x1439e6) [0x7fb390b439e6]
#25  0x7fb390b48120 in prepareProfileBased(PartDesign::Body*, Gui::Command*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<void (Part::Feature*, App::DocumentObject*)>) from /usr/lib/freecad/lib/PartDesignGui.so+0x340
#26  0x7fb390b499a2 in CmdPartDesignAdditiveLoft::activated(int) from /usr/lib/freecad/lib/PartDesignGui.so+0xf2
#27  0x7fb401904ae7 in Gui::Command::_invoke(int, bool) from /usr/lib/freecad/lib/libFreeCADGui.so+0x257
#28  0x7fb401904fdf in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad/lib/libFreeCADGui.so+0x13f
#29  /usr/lib/libQt6Core.so.6(+0x1a17e7) [0x7fb3fe7a17e7]
#30  0x7fb3ff11b0b4 in QAction::activate(QAction::ActionEvent) from /usr/lib/libQt6Gui.so.6+0x154
#31  /usr/lib/libQt6Widgets.so.6(+0x20e7d5) [0x7fb3ff80e7d5]
#32  0x7fb3ff81b22e in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/libQt6Widgets.so.6+0xde
#33  0x7fb3ff943bcd in QToolButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/libQt6Widgets.so.6+0x2d
#34  0x7fb3ff74c0ce in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0xc0e
#35  0x7fb3ff6fc55c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x9c
#36  0x7fb3ff701a01 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x10a1
#37  0x7fb401899b29 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#38  0x7fb3fe744e38 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#39  0x7fb3ff6f3e6d in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt6Widgets.so.6+0x21d
#40  /usr/lib/libQt6Widgets.so.6(+0x162072) [0x7fb3ff762072]
#41  /usr/lib/libQt6Widgets.so.6(+0x1636b0) [0x7fb3ff7636b0]
#42  0x7fb3ff6fc55c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x9c
#43  0x7fb401899b29 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#44  0x7fb3fe744e38 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#45  0x7fb3fed76cf0 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt6Gui.so.6+0x8d0
#46  0x7fb3fede5b44 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x304
#47  /usr/lib/libQt6Gui.so.6(+0x5f9594) [0x7fb3ff1f9594]
#48  /usr/lib/libglib-2.0.so.0(+0x5ca89) [0x7fb3fcf0ea89]
#49  /usr/lib/libglib-2.0.so.0(+0xbe9b7) [0x7fb3fcf709b7]
#50  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x7fb3fcf0df95]
#51  0x7fb3fe9a28dd in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#52  0x7fb3fe74f10e in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x19e
#53  0x7fb3fe74945d in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0x8d
#54  0x7fb4017a1814 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0xd34
#55  freecad(+0x7cca) [0x55821dd8bcca]
#56  /usr/lib/libc.so.6(+0x25c88) [0x7fb3fe039c88]
#57  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7fb3fe039d4c]
#58  freecad(+0x80c5) [0x55821dd8c0c5]
```

### Full version info

```shell
FreeCad compiled from git source.

OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37707 (Git)
Build type: Release
Branch: main
Hash: b463846bea5ee34bd180d0a5072f61b3da6aab44
Python 3.12.3, Qt 6.7.1, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-09-26:
could FreeCAD ship its own fork of OCCT where all these segfaults are fixed?

luzpaz, 2024-10-05:
@AIRCAP there is a more active and agreeable upstream OCCT dev that has been participating on our tracker recently (see https://dev.opencascade.org/content/occt-github-new-way-collaborate). Maybe it's our opportunity to get some bugs fixed upstream?

MatejSpindler, 2024-10-05:
[Loft.zip](https://github.com/user-attachments/files/17265668/Loft.zip)
Attached is a test case for this problem.

Steps too reproduce:
1. Select Sketch from TreeView,
2. Ctrl + Click on DatumLine in TreeView,
3. Click Aditive loft icon.

Results:
On Linux segfault as attached in the fist post. On Windows no crash, but "access violation" is printed in the report view.

From previous comments it looks like this is an upstream bug. But FreeCAD should do better.

My naive recommendation would be to ignore all Datum elements before starting the loft operation. Datum elements are infinite so loft operation with them doesn't make sense.

Hope this helps.

AIRCAP, 2024-10-18:
still crashes in latest main

stefano2734, 2026-09-12:
> still crashes in latest main

What is the state with latest occt 8.0.1 and 26.3 dev?

MatejSpindler, 2026-09-12:
I don't have the option to compile FreeCAD right now.

Using the Appimage from: [FreeCAD_weekly-2026.09.09-Linux-x86_64.AppImage](https://github.com/FreeCAD/FreeCAD/releases/download/weekly-2026.09.09/FreeCAD_weekly-2026.09.09-Linux-x86_64.AppImage)

i still get the crash. But the bundled version of occt is 7.8.1.

```
FreeCAD 26.3.0, Libs: 26.3.0devR20260908 (Git shallow)
(C) 2001-2026 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

fish: Job 1, './AppRun' terminated by signal SIGSEGV (Address boundary error)
```

**Is there an Appimage of the dev version with new occt version?**

stefano2734, 2026-09-12:
Yes, that is a bad point here. New dev version with old occt. What can you test here today with your appimage?
Nothing. 
Same old level in windows. Only in Linux latest available with libpack 3.5.5.

luzpaz, 2026-09-12:
> Is there an Appimage of the dev version with new occt version?

Maybe someone should open a ticket requesting this?

stefano2734, 2026-09-13:
> > Is there an Appimage of the dev version with new occt version?
> 
> Maybe someone should open a ticket requesting this?

Yes, two dev versions per channel would be here nice.
With occt 7.9.3 and libpack 3.4.0 3rd party level
And occt 8.0.1 and libpack 3.5.5 3rd party level Or coming occt 8.0.2 with related libpack 3.5.x 3rd party level.
So problems with transition from 7 to 8 in occt are detectable and solvable with progress in 3rd party to old releases.

pieterhijma, 2026-10-27:
I can reproduce on FreeCAD `main` with OCCT 8.1 dev:

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 27.1.0dev.49321 +1 (Git)
Build date: 2026/10/27 12:27:31
Build type: Debug
Branch: adapt-to-occt-v8_1_0
Hash: 371a550ee620d900a88d2e96bf8cb9e47b7e45ec
Python 3.14.8, Qt 6.12.0, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.1, boost 1_92, Eigen3 5.0.1, PySide 6.12.0
shiboken 6.12.0, SMESH 7.7.1.0, xerces-c 3.3.0, Clipper2 2.0.1, OCC 8.1.0.dev1-53706f78a8
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 26.2.5-arch1.1/AMD/AMD Radeon 780M Graphics (radeonsi, phoenix, ACO, DRM 3.64, 7.2.11-arch1-1)
```

In this version of OCCT, it needs a modification to a BRep file in the FreeCAD file because of https://github.com/Open-Cascade-SAS/OCCT/issues/1560.

The updated file: 

[Loft.zip](https://github.com/user-attachments/files/34020138/Loft.zip)

<details>
<summary>stacktrace:</summary>
<pre>
#0  0x00007fff93766b38 in NCollection_BaseList::Iterator::Iterator (this=0x7fffffff60f0, theList=...) at /home/pieter/scratch/dist-occt-main/include/opencascade/NCollection_BaseList.hxx:61
#1  0x00007fffaca23da3 in NCollection_BaseList::BasicIterator<NCollection_TListNode<opencascade::handle<BRep_PointRepresentation> >, opencascade::handle<BRep_PointRepresentation>, false>::BasicIterator<false, 0> (this=0x7fffffff60f0, theList=...) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKernel/NCollection/NCollection_BaseList.hxx:119
#2  0x00007fffaca223fd in NCollection_List<opencascade::handle<BRep_PointRepresentation> >::Iterator::Iterator (this=0x7fffffff60f0, theList=...) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKernel/NCollection/NCollection_List.hxx:63
#3  0x00007fff8e1b1f72 in BRep_Tool::Parameter (theV=..., theE=..., theParam=@0x7fffffff61f0: 0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/BRep/BRep_Tool.cxx:1443
#4  0x00007fff8e1b2683 in BRep_Tool::Parameter (V=..., E=...) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/BRep/BRep_Tool.cxx:1540
#5  0x00007fff915d409e in WireContinuity (W=..., contW=@0x7fffffff6700: GeomAbs_CN) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBool/BRepFill/BRepFill_CompatibleWires.cxx:344
#6  0x00007fff915d6fa0 in BRepFill_CompatibleWires::Perform (this=0x7fffffff6ab0, WithRotation=true) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBool/BRepFill/BRepFill_CompatibleWires.cxx:877
#7  0x00007fffac9937d3 in BRepOffsetAPI_ThruSections::Build (this=0x7fffffff7980) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKOffset/BRepOffsetAPI/BRepOffsetAPI_ThruSections.cxx:394
#8  0x00007fff93b0f14d in Part::TopoShape::makeElementLoft (this=0x7fffffff8580, shapes=std::vector of length 2, capacity 2 = {...}, isSolid=Part::IsSolid::notSolid, isRuled=Part::IsRuled::notRuled, isClosed=Part::IsClosed::notClosed, maxDegree=5, op=0x7fff93ccb33c "LFT") at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/Part/App/TopoShapeExpansion.cpp:4632
#9  0x00007fff86d4976b in PartDesign::Loft::execute (this=0x55555a3124a0) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/PartDesign/App/FeatureLoft.cpp:279
#10 0x00007ffff43433ce in App::DocumentObject::recompute (this=0x55555a3124a0) at /home/pieter/current/occt-liaison/freecad-occt-main/src/App/DocumentObject.cpp:166
#11 0x00007fff937555d9 in Part::Feature::recompute (this=0x55555a3124a0) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/Part/App/PartFeature.cpp:112
#12 0x00007fff86c433d3 in PartDesign::Feature::recompute (this=0x55555a3124a0) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/PartDesign/App/Feature.cpp:145
#13 0x00007ffff42c633c in App::Document::_recomputeFeature (this=0x555559917960, Feat=0x55555a3124a0) at /home/pieter/current/occt-liaison/freecad-occt-main/src/App/Document.cpp:3272
#14 0x00007ffff42c395f in App::Document::recompute (this=0x555559917960, objs=std::vector of length 0, capacity 0, force=false, hasError=0x0, options=0) at /home/pieter/current/occt-liaison/freecad-occt-main/src/App/Document.cpp:2962
#15 0x00007ffff43f41cf in App::DocumentPy::recompute (this=0x5555599246f0, args=0x7ffff51d5370 <_PyRuntime+90640>) at /home/pieter/current/occt-liaison/freecad-occt-main/src/App/DocumentPyImp.cpp:777
#16 0x00007ffff43ea855 in App::DocumentPy::staticCallback_recompute (self=0x5555599246f8, args=0x7ffff51d5370 <_PyRuntime+90640>) at /home/pieter/scratch/build-freecad-occt-main/src/App/DocumentPy.cpp:2383
#17 0x00007ffff4dd5650 in cfunction_call (func=0x7fff85b0c6d0, args=0x7ffff51d5370 <_PyRuntime+90640>, kwargs=0x0) at Objects/methodobject.c:575
#18 0x00007ffff4da59ea in _PyObject_MakeTpCall.constprop.0 (tstate=0x7ffff520c280 <_PyRuntime+315680>, callable=0x7fff85b0c6d0, args=0x7fffffff9c38, nargs=0, keywords=0x0) at Objects/call.c:242
#19 0x00007ffff4c68afd in _PyObject_VectorcallTstate (kwnames=0x0, tstate=<optimized out>, callable=<optimized out>, args=0x7fffffff9c38, nargsf=<optimized out>) at ./Include/internal/pycore_call.h:175
[... cut ...]
#82 0x00007ffff6522b4d in (anonymous namespace)::tryRunEventLoop (mainApp=...) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Gui/Application.cpp:2563
#83 0x00007ffff6522f76 in (anonymous namespace)::runEventLoop (mainApp=...) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Gui/Application.cpp:2600
#84 0x00007ffff6523516 in Gui::Application::runApplication () at /home/pieter/current/occt-liaison/freecad-occt-main/src/Gui/Application.cpp:2712
#85 0x0000555555574d0f in main (argc=1, argv=0x7fffffffd1f8) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Main/MainGui.cpp:343
</pre>
</details>

pieterhijma, 2026-10-27:
Created a test for OCCT. I have the same stacktrace:

```
#0  0x00007ffff7d7c3b4 in NCollection_BaseList::Iterator::Iterator (
    this=0x7fffffffb5e0, theList=...)
    at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKernel/NCollection/NCollection_BaseList.hxx:61
#1  0x00007ffff7a23da3 in NCollection_BaseList::BasicIterator<NCollection_TListNode<opencascade::handle<BRep_PointRepresentation> >, opencascade::handle<BRep_PointRepresentation>, false>::BasicIterator<false, 0> (this=0x7fffffffb5e0, 
    theList=...)
    at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKernel/NCollection/NCollection_BaseList.hxx:119
#2  0x00007ffff7a223fd in NCollection_List<opencascade::handle<BRep_PointRepresentation> >::Iterator::Iterator (this=0x7fffffffb5e0, theList=...)
    at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKernel/NCollection/NCollection_List.hxx:63
#3  0x00007ffff4bb1f72 in BRep_Tool::Parameter (theV=..., theE=..., 
    theParam=@0x7fffffffb6e0: 0)
    at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/BRep/BRep_Tool.cxx:1443
#4  0x00007ffff4bb2683 in BRep_Tool::Parameter (V=..., E=...)
    at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/BRep/BRep_Tool.cxx:1540
#5  0x00007ffff6fd409e in WireContinuity (W=..., 
    contW=@0x7fffffffbbf0: GeomAbs_CN)
--Type <RET> for more, q to quit, c to continue without paging--c
    at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBool/BRepFill/BRepFill_CompatibleWires.cxx:344
#6  0x00007ffff6fd6fa0 in BRepFill_CompatibleWires::Perform (this=0x7fffffffbfa0, WithRotation=true)
    at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBool/BRepFill/BRepFill_CompatibleWires.cxx:877
#7  0x00007ffff79937d3 in BRepOffsetAPI_ThruSections::Build (this=0x7fffffffcb50)
    at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKOffset/BRepOffsetAPI/BRepOffsetAPI_ThruSections.cxx:394
#8  0x000055555555b76a in main (argc=1, argv=0x7fffffffce68) at /home/pieter/current/occt-liaison/draw-test-harnasses/freecad/14640/loft/src/main.cpp:53
```

pieterhijma, 2026-10-27:
The test to reproduce: [loft.zip](https://github.com/user-attachments/files/34025276/loft.zip)

pieterhijma, 2026-10-27:
Submitted in https://github.com/Open-Cascade-SAS/OCCT/issues/1684.