FreeCAD/FreeCAD #14643: Deleting one chamfer triggers TNP

herrgerd, 2024-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hey there,
just downloaded the 0.22.0dev version because YouTube is full of "the toponaming problem seems fixed". I was hyped and so I created an example which couldn't be more basic:

[image]

I had someone else replicate it in that version, so it doesn't seem like a problem solely for me. It might be interesting to hear that this does also happen in Realthunders latest version but if you edit the second chamfer feature (e.g enter its dialog) it does find the edge again and repairs itself.

Another hint: if I made the first chamfer feature on all 4 vertical corners and deleted only 3 of them it will not destory the second chamfer feature. Only if you suppress it completely or delete it.

Cheers, Jan

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git)
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * fcgear
  * freecad.gears
  * MacroPadExtension
  * Part-o-magic
  * PieMenu
  * sheetmetal
  * ThreadProfile 1.76.0
  * workfeature
```

### Subproject(s) affected?

PartDesign

### Anything else?

[chamfer_toponaming.zip](https://github.com/user-attachments/files/15810556/chamfer_toponaming.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-13:
@bgbsww FYI

obelisk79, 2024-06-22:
Another simple test-case for chamfer. I assume is related. 

https://youtu.be/nL_o3-6u4Qs?si=4qhjpct9ry5NgVFO

Jookia, 2024-06-30:
I've been doing a little bit of investigation on this. I'm not convinced this is exactly a TNP issue. If we take the chamfer cube as created in the video like this:

[image]

Then update each chamfer:

[image]

Everything works fine. Suppressing however doesn't work very well, while it does in LinkStage3.

This seems to be an issue of FreeCAD not automatically setting a new object when one is deleted.

Jookia, 2024-06-30:
Huh. This actually works if you suppress the first chamfer before deletion!

bgbsww, 2024-07-04:
Is this actually TNP, or does the title and tagging need adjustment?

This appears from the descriptions to be a chamfer bug that occurs both in main and LS3.

Do we have a sense of how it works on the .21 release?

I'll take a poke around the suppressing stuff to look for obvious things.  It would be helpful to see some console logging that shows missing references or the like to confirm that it is TNP related.

maxwxyz, 2024-07-07:
@herrgerd could you please test in the stable 0.21.2 version?

herrgerd, 2024-07-08:
> @herrgerd could you please test in the stable 0.21.2 version?

[image]

herrgerd, 2024-07-08:
> I've been doing a little bit of investigation on this. I'm not convinced this is exactly a TNP issue. If we take the chamfer cube as created in the video like this:
> 
> ![A cube with two chamfered edges created by two separate chamfers](https://private-user-images.githubusercontent.com/185275/344440356-1a8e1761-f929-4134-a1d7-ade1bfc7660a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA0NTc0MjAsIm5iZiI6MTcyMDQ1NzEyMCwicGF0aCI6Ii8xODUyNzUvMzQ0NDQwMzU2LTFhOGUxNzYxLWY5MjktNDEzNC1hMWQ3LWFkZTFiZmM3NjYwYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwOFQxNjQ1MjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iYWRmYTNkYjFlOGIyN2RhYzEyMDc2NWIxMmZlZDE5NmFhZTlhZGJiYTZhOTA0NjQwOTg1YjYzNjk3NjVmZTM5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.C9heBsKQrTSC-Mr_s_tCvmS8BI4jYQI0CXti0HQ4hww)
> 
> Then update each chamfer:
> 
> ![The same cube and two chamfers, but with more edges selected](https://private-user-images.githubusercontent.com/185275/344440372-9224309b-3a4a-4f5f-8aaa-fb15f654e57b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA0NTc0MjAsIm5iZiI6MTcyMDQ1NzEyMCwicGF0aCI6Ii8xODUyNzUvMzQ0NDQwMzcyLTkyMjQzMDliLTNhNGEtNGY1Zi04YWFhLWZiMTVmNjU0ZTU3Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwOFQxNjQ1MjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNmFlOWMxMzMyNjQ0OTA2ZGFmNjI3NDZiOThmMjEzYWMyNDljNTg2MzlmZTRlZjY2ZGY3ZjFkNGMzNTAxYmFlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.DnPRNBcjKtRLKH0rmNglh-prcr6su5C9E4ypCiW9GOY)
> 
> Everything works fine. Suppressing however doesn't work very well, while it does in LinkStage3.
> 
> This seems to be an issue of FreeCAD not automatically setting a new object when one is deleted.

**I re-did it in 22DEV in fullscreen plus I added the console output:**
If the GIF is too small right-klick and load it in an extra tab.

[image]

Console output:
19:12:46  <PropertyLinks> PropertyLinks.cpp(455): Unnamed#Chamfer001.Base missing element reference Unnamed#Chamfer ;g144;SKT;:H8bb,E;:G;XTR;:H8bb:7,F;:U;XTR;:H8bb:7,E;:M;CHF;:H8bc:7,E.Edge13
19:12:46  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ;g144;SKT;:H8bb,E;:G;XTR;:H8bb:7,F;:U;XTR;:H8bb:7,E;:M;CHF;:H8bc:7,E.Edge13
19:12:46  Chamfer001: Invalid edge link: ;g144;SKT;:H8bb,E;:G;XTR;:H8bb:7,F;:U;XTR;:H8bb:7,E;:M;CHF;:H8bc:7,E.Edge13
19:12:53  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ?Edge13
19:12:53  Chamfer001: Invalid edge link: ?Edge13
19:12:57  <PropertyLinks> PropertyLinks.cpp(455): Unnamed#Chamfer001.Base missing element reference Unnamed#Chamfer ;g144;SKT;:H8bb,E;:G;XTR;:H8bb:7,F;:U;XTR;:H8bb:7,E;:M;CHF;:H8bc:7,E.Edge13
19:12:57  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ;g144;SKT;:H8bb,E;:G;XTR;:H8bb:7,F;:U;XTR;:H8bb:7,E;:M;CHF;:H8bc:7,E.Edge13
19:12:57  Chamfer001: Invalid edge link: ;g144;SKT;:H8bb,E;:G;XTR;:H8bb:7,F;:U;XTR;:H8bb:7,E;:M;CHF;:H8bc:7,E.Edge13
19:13:05  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ;g144;SKT;:H8bb,E;:G;XTR;:H8bb:7,F;:U;XTR;:H8bb:7,E;:M;CHF;:H8bc:7,E.Edge13
19:13:05  Chamfer001: Invalid edge link: ;g144;SKT;:H8bb,E;:G;XTR;:H8bb:7,F;:U;XTR;:H8bb:7,E;:M;CHF;:H8bc:7,E.Edge13
19:13:06  Editing object: Unnamed#Chamfer001
Object path: Unnamed#Body.Chamfer001.
Editing element: 
Edit mode: 0
19:13:09  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ?Edge13
19:13:09  Chamfer001: Invalid edge link: ?Edge13

bgbsww, 2024-07-08:
Thanks, that console log is very helpful.

bgbsww, 2024-07-08:
After much toying with this both in main and LS3, this is consistent  so perhaps not a blocker.

However, the related question, which is one of repairing the broken link like LS3 does is worth continuing to look into and might be considered a blocking issue.

bgbsww, 2024-07-09:
Also #15054 is a better working case for the suppression bug - which is completely missing code, so this one is probably best focused on just chamfer, which I think isn't an issue anymore.

maxwxyz, 2024-07-10:
@bgbsww should this issue be closed then?

bgbsww, 2024-07-10:
I think so; I was just waiting to see if there was any other example raised, and that doesn't appear to be happening.

herrgerd, 2024-07-10:
So, what's the solution here? Is the bug adressed or is it just closed? I'm not a dev so I am not sure how to interpret things :)

bgbsww, 2024-07-10:
There are two issues raised here.  One is the suppression issue which was addressed in the patch for #15054.  The other is the labeled "Deleting one chamfer triggers TNP".  Since everything works the same in LS3 and in main, I don't think this is a blocker, and I don't intend to attempt to fix it.  It isn't so much triggering TNP as identifying that the TNP mitigations do not try to guess the right answer here, or present a choice to the user.

So, either this should be closed as "not a bug", or it should be relabeled as something like  "FreeCAD not automatically setting a new object when one is deleted", not classified as a blocker, and not assigned.

I'm trying not to arbitrarily kill what looks like a legitimate issue to get investigated in the future, while holding the line on blockers that are actually new work hiding in bug reports.

QuinFromCanada, 2024-07-10:
Chamfers, fillets and drafts seem to still move to different edges on complex geometries, the funny part is that they often show up on the same place on the preview but when you go into selection mode the line can be at a different location.   I have a big design that I can link if you want to give it a try, I infact I've even ran into a situation where changing sketch will cause a chamfer to move, fixing it will break a other chamfer, and when you get to the last chamfer it will break the first chamfer.  You can go around in circles doing this.

Now that being said, I originally did this design in the RT branch, when I had issues with it in dev 0.22 I remade it from scratch thinking it version issue, but when I recreated it I saw the same issue.

I can post a video if you would like,

Hope this helps, keep up the great work, I'm loving the release.

bgbsww, 2024-07-10:
Thanks!  You are confirming that there is some subtle problem here that isn't visible yet.  The videos aren't super useful unless I don't believe you, and I do.  (Herrgerd I believe you too, just need something I can grab onto). Can you confirm that you're seeing similar log messages to above ( missing element references and Invalid Edge links ) ?

I suspect the eventual solution to this will come from unit tests on every permutation until we find what is triggering the problem.  I will eventually get to that, when there aren't directly solvable bugs clamoring for attention.

If someone wants to try to turn the initially reported example into tests that can be shown to work in LS3 and break on main, then we are more than half way there; again a step I'll take eventually, but that could be addressed in the interim.

QuinFromCanada, 2024-07-11:
@bgbsww yes I am getting similar console outputs.

If it helps you, I've uploaded the file to GIT [https://github.com/QuinFromCanada/FC_ErrorReporting](https://github.com/QuinFromCanada/FC_ErrorReporting)  File is called cage.FCStd

If you open the file, 
1. Hide the last 2 features (**mirror** and **Pocket** just to make it easier).
2. Look for **chaTopOfCrossbar** (2/3s down the tree). Edit the chamfer and observed the location with the select button.
3. The feature just above it is **chaPillarTopInsideStrength** delete it.  you'll notice that the console will output:

```
20:48:31  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ;Edge79;:Hbb8,E;:M;CHF;:Hbb9:7,E.Edge57
20:48:31  chaTopOfCrossbar: Invalid edge link: ;Edge79;:Hbb8,E;:M;CHF;:Hbb9:7,E.Edge57
```
**chaTopOfCrossbar** will how an error, BUT the model will still show the same.  ie the feature we deleted (chaPillarTopInsideStrength) and the feature that's in error still show on the 3d model.

4. Edit **chaTopOfCrossbar** again and observe the location with the select button.  the location of the edges has changed however, the edges that it highlighted are not actually chamfered when you go back to preview mode.

Now sometimes when you edit other features the edge location of some other chamfers will be lost and will display as **?Edge#** when you edit the chamfer, BUT it will not have a red Error icon on the feature and it will still show up on the 3D view.  I'll post the instructions for this if I figure out how to replicate it reliably.

Secondly, my LS3 design of this design is 2 major revisions behind. If I delete a similar chamfer I will get the following in the console:

```
21:14:19  <Exception> FeatureDressUp.cpp(140): Invalid edge link: Edge14
21:14:19  Recompute failed!

```
The next chamfer down in the tree will show and error but, when I go into that chamfer its location of the edges is still the same and exiting the feature will make the error icon disappear.  If it helps I can bring bring the LS3 version inline to the 0.22dev revision and experiment with it more.

Hope this helps.

wsteffe, 2024-09-24:
I do not think that error triggered by suppression of first chamfer in the cube is a bug.

I have made the cube with 2 chamfers in current LS3 stable.
As said by [QuinFromCanada](https://github.com/QuinFromCanada) after suppressing firts chamfer there, an error marker appears in the second chamfer. After opening it "?Edge13" is reported in the edge list.
Edge13 is the correct edge (has not changed) but the question mark (in my interpretation) means that TPN code is not sure of this guess
and asks for user confirmation. The confirmation is done selecting the edge in the list and closing the task. Then the Error disappears.

I think that this is the correct behaviour when TNP guess is not supported by a strong evidence.  TNP code can not be sure that Edge number 13 is the same edge. The numbering could have changed after suppression of the first chamfer.

luzpaz, 2024-09-24:
Renamed ticket per @bgbsww

wsteffe, 2024-09-24:
[QuinFromCanada](https://github.com/QuinFromCanada) wrote:
`when you get to the last chamfer it will break the first chamfer.`

I think that when you are saying the last chamfer you in mean the last as appearing in the structure tree.
This circular recursion should never happen if your changes are ordered according to the dependancy graph which should always be DAG (Direct Acyclic Graph). With this kind of graph it should always be possible to order the changes in such a way that previous changes cant't be affected by subsequent changes. The good ordering is not unique and, unfortunately, it is not shown in the structure tree.

luzpaz, 2024-10-17:
@CalligaroV can you keep an eye on this ticket if you have the bandwidth ?

CalligaroV, 2024-10-18:
@luzpaz yes, sure!

Unfortunately my bandwidth is pretty narrow as I'm following other issues/PRs that need me to study quite a bit of the TNP code, therefor ATM I can't guarantee I'll be able to do a good analysis / give useful feedback / open a PR with possible fix in short time.

By looking at the screenshots to me this looks basically about the code behind the `Suppress` function, which I don't know well enough.

I remember however that I played a little bit with that part of the code while trying to find a fix for another issue (or maybe this one)?
I'll need some time to reconnect my thoughts :confounded:

Is this still a 1.0 Milestone?

luzpaz, 2024-10-18:
Thank you. Yes, I figured you had a lot going on and wanted to respect your availability. Thank you for weighing-in. 

> Is this still a 1.0 Milestone?

It still is AFAICT.

maxwxyz, 2025-03-29:
This not also happens when suppressing or deleting features but happens all the time when I slightly change a sketch or pad length which does not introduce new edges or geometry, so simply slight modifications do break the model whcih did not happen in 0.21 as the edge and face names are still the same.
Here is a screenshot of the issue, where the original edges in the fillet dialog are displayed with `?Edge`. I manually need to add the same edges so they appear without a question mark and manually delete the ones with a question mark to get it working. Unfortunately on every edit:

[image]

@kadet1090 was https://github.com/FreeCAD/FreeCAD/pull/17768 intended to solve this?