FreeCAD/FreeCAD #14645: [Problem] On-View-Parameters indication on circle is confusing

MiqCG, 2024-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With On-View-Parameters active (Edit > Preferences > Sketcher > General > Tool Parameters) and "Dimension tool diameter/radius mode" on Diameter or Auto, a circle drawn in Sketcher shows an arrow indication (with two "sharp ends") but with one end pointing to the center of the circle. I think this is confusing, as look like is asking for the radius of the circle. See Screenshot_1.

When "Dimension tool diameter/radius mode" is on Radius, the dimension indication have only one sharp end. See Screenshot_2.

I have searched in Github [is:issue circle sketcher dimension radius] and [is:issue circle sketcher dimension diameter] with no relevant results.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git) AppImage
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * BillOfMaterials 0.0.8.5
  * fasteners 0.5.22
```

### Subproject(s) affected?

Sketcher

### Anything else?

[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-13:
@PaddleStroke / @FlachyJoe / @howie-j  FYI

PaddleStroke, 2024-06-13:
Yes there is a problem here. I'll fix this

PaddleStroke, 2024-06-13:
Ah yes this was introduced by https://github.com/FreeCAD/FreeCAD/issues/12384

closed 2024-06-13 by chennes