FreeCAD/FreeCAD #14647: [Problem] Alphabetically sorting option of workbenches (Preferences) is difficult to discover

MiqCG, 2024-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Within Edit > Preferences > Workbenches > Available Workbences, in the text of the upper section is advised that the WB can be reordered by dragging but there is no mention about the possibility of sorting them alphabetically. 
I think that finding how to do it (by right clicking inside the list) by oneself is unlikely. 

To improve that, I have two proposals:
1) add some text in the upper area that explains the possibility to sort alphabetically and how to do it, or

2) add a checkbox on the bottom area, with "Sort workbenches alphabetically.

I find more evident and easy the second proposal, although I suppose is harder to implement as the checkbox will have to automatically uncheck when a workbench is dragged after alphabetical sorting.

I have searched in Github [is:issue alphabetically] with no relevant results.

## Subproject(s) affected?
Preferences and UI/UX

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git) AppImage
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * BillOfMaterials 0.0.8.5
  * fasteners 0.5.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-13:
A checkbox is not good in this case. You just need a button.
I'm not sure of the value of having a button for this though. That's for the @FreeCAD/design-working-group to weight in.

MiqCG, 2024-06-13:
Thanks
Ping @obelisk79 to amend the Design Working Group unsuccessful ping above.

Syres916, 2024-06-13:
With regard to proposal 1, would this be acceptable:

[image]

closed 2024-07-01 by chennes