FreeCAD/FreeCAD #14649: If Main window is not Maximixed, exiting FreeCAD and restarting, the Main window will not restore to the prior position

Buffalo1John, 2024-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you start FreeCAD, Restore Down the main Window, move it on the screen , then exit the program, upon restart the main window will not return to the position it was in when exiting occured.

It works correctly on 0.21.2

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git)
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.64a
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-13:
It should've been fixed with https://github.com/FreeCAD/FreeCAD/pull/11275#issuecomment-1805231598
@wwmayer FYI

Buffalo1John, 2024-06-13:
I just tried it on 0.21.2:
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.64a
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * QuickMeasure 2022.10.28
[/code]

closed this:
[image]

when opened saw this:
[image]

Buffalo1John, 2024-06-13:
Here it is with 0.22dev:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.64a
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * QuickMeasure 2022.10.28

closed this:
[image]

then opened to see this:
[image]

maxwxyz, 2024-07-07:
is it still an issue in the latest weekly?

Buffalo1John, 2024-07-07:
Yes, this is still an issue...

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.65
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * QuickMeasure 2022.10.28

Buffalo1John, 2024-08-09:
Still found on 0.22.0dev.38419

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.65
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * QuickMeasure 2022.10.28

luzpaz, 2024-10-17:
Related: #17221

Buffalo1John, 2025-07-04:
This remains a bug in weekly build 42461 which I just downloaded and tested.

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42461 (Git)
Build date: 2025/06/30 16:12:02
Build type: Release
Branch: main
Hash: 60aa5ff3730d77037ffad0c77ba96b99ef0c7df3
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * A2plus 0.4.68
  * Curves 0.6.61
  * fasteners 0.5.38
  * lattice2 1.0.0

maxwxyz, 2025-09-24:
Same happens for me on Win11

ipatch, 2025-11-09:
is this just something specific to windows 11 or does it affect all OSes ie. gnu+linux and macos as well?

ipatch, 2025-11-10:
unfortunately i use rdp (via X11vnc (don't ask 😑)) to login to this m1 mac running asahi (it doesn't have a monitor connected to it). so the dekstop is just one screen / main window. so i can not test across multiple monitors / screen boundaries that i know of.

luzpaz, 2025-11-28:
@Syres916 i think I saw you comment on a similar ticket recently regarding screen coordinates... or am I totally wrong ?