FreeCAD/FreeCAD #14650: When Undocked, the Tasks window will shift down and right

Buffalo1John, 2024-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have undocked the Model and Tasks windows on my multiple monitor setup. Please NOTE that even if I operate with only one monitor on my laptop, this behaviour described here occurs. These two windows are to the left of the main FreeCad window. If I start up a clean session, I first make sure the two windows mentioned are lined up on their tops. Then, when I do a variety of things, only the Tasks window will shift down and to the right. It did this in the prior two dev builds as well. 

It does not do this in 0.21.2

I will refer to this shifting behaviour as walking in the description below. The Model window does not walk at all. If the Model window is docked, the Tasks window still walks, so the Model window is irrelevant to the problem (I only mention it as a window that does not move when undocked.

Here is an example:
click **New** (no walking)
click **Body** (no walking)
click **Sketch** (Tasks - walk 31 pixels down, 1 pixel right)
click **XY plane** (Tasks - walk 31 pixels down, 1 pixel right)
draw stuff - doesn't matter what it is, make a circle or a rectangle for padding (no walking)
click **Close** (Tasks - walk 31 pixels down, 1 pixel right)
click **Pad** (Tasks - walk 31 pixels down, 1 pixel right)
click **Cancel** (Tasks - walk 31 pixels down, 1 pixel right)

If you go back into the Sketcher, Tasks walks going in and closing
if you pad the first sketch and then create another sketch on the face of the pad, walking does same things, though now you can pocket and Tasks walks as well, so it is not limited to sketch and pad.

If you continue going in and out of the Sketcher (or do Pad then Cancel), you can walk the Tasks window so it almost disappears from the screen. Yes, this extreme, but it doesn't take long before, when doing normal design work, that you have to move the Tasks window back into usable space...

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.64a
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]
[image]
[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Buffalo1John, 2024-07-07:
This is still an issue on the latest build

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.65
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * QuickMeasure 2022.10.28

luzpaz, 2024-07-21:
@MisterMakerNL have you encountered this ?

Buffalo1John, 2024-08-09:
Still found on 0.22.0dev.38419

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.65
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * QuickMeasure 2022.10.28

Trallala-Hobsassa, 2024-09-30:
Hi!

This happen on 1.0.0RC2 too. And the Panels don't rember their Positions when Freecad was closed and startet again....

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * fasteners 0.5.29
  * freecad.gears 1.3.0

MisterMakerNL, 2024-09-30:
I never work like this, but I tested it and can confirm this!
This shift happens every-time the task window gets reset/renewed.

Buffalo1John, 2024-09-30:
I work like this because I have two monitors in addition to my laptop. I put the main window on one monitor and then the other windows I find useful (Model, Report View, and Tasks) on the other monitor and the laptop screen can be used to put drawings or other stuff for reference when designing. 

A workaround for now, since my monitors are bumped up against each other, is to make the main window cross over to the left monitor so the Tasks panel docked to the main window actually displays on the left monitor. The problem with this is the menu and docked tool bars are crossing the window boundary, so they are kind of a pain to deal with. I really don't like this kludgy workaround. Here is what it looks like, the black band being between the two monitors (and this is scaled to the actual distance of the frames of the monitors) and the green band being between the middle screen to the laptop.
[image]

MisterMakerNL, 2024-09-30:
@Buffalo1John This a real bug, if we can undock the panels they shouldn't move, so thanks for reporting!

Buffalo1John, 2024-10-16:
This bug is also in 1.1.0dev, just in case anyone was wondering...

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
  * lattice2 1.0.0
  * ThreadProfile 1.92.0

Buffalo1John, 2024-10-16:
> Hi!
> 
> This happen on 1.0.0RC2 too. And the Panels don't rember their Positions when Freecad was closed and startet again....
> 

I reported this other above bug  about remembering positions as:
UX: If Main window is not Maximized, exiting FreeCAD and restarting, the Main window will not restore to the prior position
https://github.com/FreeCAD/FreeCAD/issues/14649

That would be the bug report number just before this bug bug report's number - #14650

LupinIII, 2024-12-13:
Is there any progress on this? Any other workarounds in the meantime? I can't use "extend the main window over two screens and keep tasks docked" as my monitors are different resolutions (the side monitor is smaller) and in addition I have them vertically offset on purpose by about 30 pixels (side monitor being lower) so that the top right corner of my main monitor "catches" the mouse (to easily hit the "close window" icon of maximized windows).

LupinIII, 2024-12-16:
I was so annoyed by this bug that I wrote a workaround that keeps the undocked "Tasks" window in place. Unfortunately it's not a FreeCAD patch but an AutoHotkey v2 script. 
Start this script *before* you start FreeCAD. It listens to window creation events and won't pick up already open windows. It shouldn't mess anything else up with Windows, but no guarantees. I couldn't get it to trigger on the windows close event for example (when closing FreeCAD) for cleanup, so it listens for move events of a non-existing window forever. It will still pick up the next "Tasks" windows of following FreeCAD sessions. But I suggest you restart this script after you've started FreeCAD about a thousand times :wink: .
You might have to adjust some parts of the script depending on your system (like window title to search for, the window class, the pixels that the window moves). If you want to see some debug output, run Microsoft's DebugView (it's a 1MB portable by Microsoft Sysinternals).
You also have to download the library file https://github.com/Descolada/AHK-v2-libraries/blob/main/Lib/WinEvent.ahk (just match the #include to where ever you put it).

```
#Requires AutoHotkey v2
#include .\lib\WinEvent.ahk

DetectHiddenWindows True

;-- Listen to Window Create events; filter by exe and window class; class found with AutoHotkey Window Spy
WinEvent.Create((hWnd, hook, dwmsEventTime) => WindowLocker(hWnd), "ahk_exe freecad.exe ahk_class Qt51515QWindowToolSaveBits")

class WindowLocker {
    __New(hWnd) {
        this.hWnd := hWnd
        this.countdown := 5
        this.x := 0
        this.y := 0
        this.w := 0
        this.h := 0
        SetTimer ObjBindMethod(this, "CheckForTasksWindow"), 500
    }

    __Delete() {
        OutputDebug "[D](AHK) " this.hWnd " destructed"
    }

    CheckForTasksWindow() {
        ;-- sometimes title isn't set to "Tasks" right on window creation so try multiple times
        if (this.countdown > 0) {
            if (WinExist(this.hWnd)) {
                if (WinGetTitle(this.hWnd) == "Tasks") {
                    OutputDebug "[I](AHK) Tasks Window found"
                    SetTimer , 0
                    WinGetClientPos &x, &y, &w, &h, this.hWnd
                    this.x := x
                    this.y := y
                    this.w := w
                    this.h := h
                    OutputDebug "[I](AHK) Tasks Window position: (" this.x "),(" this.y ")"
                    ;-- listen to move events on "Tasks" window
                    WinEvent.Move(ObjBindMethod(this, "TasksWindowMove"), this.hWnd)
                    WinEvent.Close(ObjBindMethod(this, "TasksWindowClose"), this.hWnd)
                } else {
                    OutputDebug "[D](AHK) " this.hWnd " " WinGetClass(this.hWnd) " " WinGetTitle(this.hWnd)
                    this.countdown--
                }
            } else {
                OutputDebug "[W](AHK) " this.hWnd " no longer exists"
                SetTimer , 0
            }
        } else {
            OutputDebug "[D](AHK) " this.hWnd " counted to 0"
            SetTimer , 0
        }
    }

    TasksWindowMove(hWnd, hook, dwmsEventTime) {
        ;OutputDebug "[D](AHK) event hook EventType: " hook.EventType
        WinGetClientPos &x, &y, &w, &h, this.hWnd
        if (x == this.x and y == this.y) {
            OutputDebug "[D](AHK) Tasks Window move event without coordinate change. Ignoring"
            return
        }
        OutputDebug "[D](AHK) Tasks Window moved from (" this.x "," this.y ") to (" x "," y ")"
        if ((Abs(x - this.x) == 1 or Abs(x - this.x) == 8) and Abs(y - this.y) == 31) {
            ;-- buggy window move is either 1px right and 31px down or 8px left/right and 31px up/down
            OutputDebug "[I](AHK) Tasks Window moved by bug. Undoing"
            WinMove this.x, this.y, this.w, this.h, this.hWnd
        } else {
            ;-- every other movement is most likely due to user moving the window, so update the "locked" position
            OutputDebug "[D](AHK) Tasks Window moved by mouse"
            this.x := x
            this.y := y
            this.w := w
            this.h := h
        }
    }

    TasksWindowClose(*) {
        ;-- could not get this to trigger
        OutputDebug "[I](AHK) Tasks window closed"
    }
}
```

Buffalo1John, 2024-12-17:
**I don't think this is a valid workaround**. 

I can't use it. I don't have the first clue what to do with the "script", even though it has some instructions. Does it work for Windows 10 build 19045? Does it work for Windows 11? 

The information I could find was that DebugView isn't what Win10 needs, it is something else that is found in the Windows SDK (according to a stackoverflow post from 2015) and I don't have a Windows SDK.

Are we going to classify real bugs with workarounds that require us to run scripts that are not part of FreeCAD?

Buffalo1John, 2024-12-17:
Also for the supposed workaround - does it work for Linux and Mac and Arm? Don't they all use QT?

xtemp09, 2024-12-17:
> Also for the supposed workaround - does it work for Linux and Mac and Arm? Don't they all use QT?

_FreeCAD_ on _Linux_ (at least _KUbuntu_) does not have this bug.

xtemp09, 2024-12-17:
> My guess is that FreeCAD "draws" contents into a normal window and then tries to shift the content to "hide" the normal window title bar

I bet $1 on hardcoded margins in the code or QSS.

Trallala-Hobsassa, 2024-12-17:
Hello!

 

I just wanted to say thank you! And that I am grateful to you for the workaround! 

 

I changed the window title in the script to German, put the library in a subfolder "lib" of the directory where the script is located and installed AutoHotKey.exe and it works great. 

 

I made a batch file that first starts the script and then FreeCAD and I was happy!

 

Thank you again for the workaroud!

Best Regards
Trallala! 

 

For me it is a workaround. Should it be the norm that FreeCAD has to rely on external programs to work properly? Definitely NOT!

I was thinking about posting it here since GitHub "issues" should not be used like forum threads, but albeit small this is currently THE most annoying FreeCAD bug for me. That is why I share my workaround here (as in a different case I'd maybe be one to use someone else's workaround even if it's something external, and I'd be thankful for that).

I have no way to test on Linux, but my assumption was (which is apparently correct) that it only affects Windows. I'm (unfortunately) used to QT being somewhat flakey on Windows. FreeCAD does something to override Windows' UI look for its sub-panel windows. The offset that is introduced every time a task is opened or closed (1px right, 31px down) is exactly where the top left corner of the content of a Windows window is (so window minus titlebar and window border), at least on my system. That's why I wrote the pixel values might have to be adjusted if you use a different theme on Windows.

My guess is that FreeCAD "draws" contents into a normal window and then tries to shift the content to "hide" the normal window title bar. But something goes wrong there which leads to this bug and the window moving instead.

Further explanation:
AutoHotkey ( https://www.autohotkey.com/ ) is an open source interpreter/scripting language for Windows (all versions) that allows you to "script" the Windows UI (listen to events like hotkeys and window events, press "buttons", move windows, and much, much more). Save the above script in a file with the extension ".ahk" (don't forget to to also get the above mentioned library file). Run the script by double clicking it.
DebugView (the very first Google search result: https://learn.microsoft.com/en-us/sysinternals/downloads/debugview ; I don't know what you found, but 10 year old stackoverflow posts shouldn't be your starting point) is not needed to run the script. You use it if you need to debug an AutoHotkey script while writing/changing it. DebugView is to AutoHotkey what the "Report View" panel is to FreeCAD, not necessary in normal use, but invaluable when developing.

Buffalo1John, 2024-12-17:
Can you guys take this to the forum?

If we could get more input on where this fails and by what amounts, that would build more to help fix it. I don't recall seeing the 31 pixel jump, 

I'm sure there are much smarter folks and maybe someone who knows more about the quirks and guts of FreeCAD, QT, and perhaps QSS (I don't know what that is) will look at this and say "Oh, I know what that is". To me it looks like somebody saved the state of the window, entered the task, clobbered something, exited the task and it isn't seen because the window is docked or not windows or something else like that.

I don't object to there being a way to work around this bug, I just don't want to see this bug to be ignored. I had hoped when I reported it half a year ago that it would make it into 1.0, now it is labled 1.1, and I fear it will be labeled closed as there is a "workaround". If you can get this on the forum, we can also judge how many more people want this fixed.

LupinIII, 2025-08-28:
Yes, it seems like I shouldn't have used the word "workaround" for my script. I don't know if that triggered the tag "has workaround", but if I had known that that tag means "won't be assigned to anyone in 8 month", I wouldn't have (might not even have shared the script). This bug still annoys me even with my script, because I often forget to start my script before FreeCAD, which means it won't work (because I didn't program my script that way).

I hope this comment bumps the bug report a bit more and someone removes the "has workaround". IMHO only problems that can be "worked around" within FreeCAD merit that tag.

Buffalo1John, 2025-08-28:
Thanks for you reply!

I'd donate money to the bug fix project if someone would take care of this one - I first reported it on June 12, 2024... 

Maybe a little late for the upcoming code freeze... 

I know it is not as important as some others and there are still many other bugs that have not been tackled. I wish I knew how to get to the relevant code - maybe I could look at it and find something obvious, but alas, if I found anything, I wouldn't know how to get someone to do anything about it....

MisterMakerNL, 2025-08-28:
I did some digging in this and the task panel does not move. But the widgets in it... It shifts down with 58 pixels every-time.

Buffalo1John, 2026-03-08:
This is still a problem, tested against 1.2dev - still the very same problem. 

This is actually a serious problem in that if you don't keep manually reseting the position of the tasks window, it can be hard to grab it to move it back up.