FreeCAD/FreeCAD #14651: assembly  not when adding the first component it does not hook to the origin

Vincenzoferrara, 2024-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

when i create a new assembly, and import it, it should behave in the origin (at least the first one)

https://github.com/FreeCAD/FreeCAD/assets/64591523/2f0bc2c7-0e1f-4d2a-86bc-635206a26a73

This is useful to understand if once assembled all the angles are correct

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git) AppImage
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT) [ OS: Italian/Italy (it_IT) ]
Installed mods: 
  * AutoOpenSpreadsheet
  * ProDarkThemePreferencePack 1.0.0
  * CommandPanel
  * IconThemes
  * fasteners.backup1715436808.37841 0.5.16 (Disabled)
  * FeedsAndSpeeds 0.5.0
  * ExplodedAssembly
  * OpenTheme 2024.5.16
  * Freecad-Built-in-themes-beta 1.2.2
  * fasteners 0.5.21
  * TabBar
  * Assembly4 0.50.12
  * CubeMenu
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-06-13:
I agree that there should be a way to snap the first part to the origin but the user should be able to move it elsewhere if needed.

Vincenzoferrara, 2024-06-13:
When it is created it should ask if you want to generate in the source.

First it asks if you want to set it as background

pierreporte, 2024-06-13:
@Vincenzoferrara I don’t understand your last comment.

maxwxyz, 2024-06-13:
https://github.com/FreeCAD/FreeCAD/issues/14575#issuecomment-2161303754

sheetmetalman, 2024-07-01:
I think it could be done if the origins planes of the part are coincident with origin planes of the assembly. This would be great!

Futhermore I think it should be possible to make joints (like distante, paralell, perpendicular etc.) with the datum planes.

PaddleStroke, 2024-07-04:
The parts are not added at the origin, this was a feature request, that instead of adding to the origin, the part is inserted in the center of the screen. So that if you are far from the origin the part is still inserted where you can see it.

I can optimize this to check if the origin is within the view frame, and if it is use it. Else use the center of the screen.

pierreporte, 2024-07-04:
There could be an option to insert components at the original by default. When building an assembly from scratch, it is often desired to have the first part at the origin. This is not always true, hence the options. Next parts are rarely wanted at the origin though, but it can happen. The option could be a checkbox in the component insertion panel, and it's default state a program setting.

maxwxyz, 2024-07-04:
Yes, I also would prefer only the first part to be at the origin (origins aligned, not COM at origin). Checkbox would be good: "Align origins".

pierreporte, 2024-07-04:
Agree. First part at origin by default, next ones are free. By default, the checkbox is checked when there is no part yet, and unchecked after the first part is inserted. In the settings, two checkboxes are necessary: one for the first part (at origin or free, default is origin), one for the next parts (at origin or free, free is default). Alignment with main axis is important as well.

PaddleStroke, 2024-07-05:
What about the solution I offered : 
- If the origin is in the view : insert at origin
- If the origin is not in the view : insert at the middle of the screen.

I think it would work in almost all cases, since usually when you create an assembly and insert the first part, the origin will be on the screen.

This way we avoid introducing 2 preferences.

(Also currently we have UI feature freeze. So my current PR can be accepted as bug fix but I can't start adding checkboxes and preferences.)

maxwxyz, 2024-07-05:
@PaddleStroke is this only valid for the first part or all?

Vincenzoferrara, 2024-07-08:
yes also in my opinion only the first object should be inserted at the origin

closed 2024-07-14 by chennes