FreeCAD/FreeCAD #14654: Exception TopoShapeExpansion.cpp(2036): Invalid shape name

szimszon, 2024-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I try to select MiddleDiskPocketSketch FreeCAD crashes:

[image]

with:
```
<Exception> TopoShapeExpansion.cpp(2036): Invalid shape name ?Edge9
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42990) [0x713fcb042990]
#1  0x713f2d1170a8 in SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const> > > const&, bool) from /tmp/.mount_FreeCAdODja2/usr/lib/SketcherGui.so+0x198
#2  0x713f2d10e24b in SketcherGui::EditModeCoinManager::updateColor(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const> > > const&) from /tmp/.mount_FreeCAdODja2/usr/lib/SketcherGui.so+0x3b
#3  0x713f2d10e2f4 in SketcherGui::EditModeCoinManager::updateColor() from /tmp/.mount_FreeCAdODja2/usr/lib/SketcherGui.so+0x34
#4  0x713f2d138a3d in SketcherGui::ViewProviderSketch::onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAdODja2/usr/lib/SketcherGui.so+0x23d
#5  0x713fce8df2dc in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x2c
#6  /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so(+0x782c6d) [0x713fce582c6d]
#7  0x713fce8e8f97 in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x1e7
#8  /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so(+0x782c6d) [0x713fce582c6d]
#9  0x713fce8e0cce in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x1be
#10  0x713fce8e1445 in Gui::SelectionSingleton::rmvPreselect(bool) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x3d5
#11  0x713fce62bb91 in Gui::SoFCUnifiedSelection::setHighlight(SoFullPath*, SoDetail const*, Gui::ViewProviderDocumentObject*, char const*, float, float, float) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x181
#12  0x713fce633087 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x2a7
#13  0x713fca8c5e48 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/./libCoin.so.80+0x68
#14  0x713fcaa7fcba in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/./libCoin.so.80+0x15a
#15  0x713fcaacfcd8 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/./libCoin.so.80+0x58
#16  0x713fcaaeaf7a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/./libCoin.so.80+0x2a
#17  0x713fca8c5e48 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/./libCoin.so.80+0x68
#18  0x713fca8cfa3c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/./libCoin.so.80+0xac
#19  0x713fca8c6b2a in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/./libCoin.so.80+0x2ea
#20  0x713fcaa9bdc2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/./libCoin.so.80+0x72
#21  0x713fce6e001e in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x1e
#22  0x713fce6eb3f8 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x58
#23  0x713fce6f76d0 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x40
#24  0x713fce6fdeff in Gui::TouchpadNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x2df
#25  0x713fce6f7965 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x65
#26  0x713fce71fe5b in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x6b
#27  0x713fce6d66f5 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x1e5
#28  0x713fcb6b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Core.so.5+0x7b
#29  0x713fcc57039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#30  0x713fcc5778f9 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Widgets.so.5+0xa39
#31  0x713fce337629 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x99
#32  0x713fcb6b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Core.so.5+0x108
#33  0x713fcc5763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#34  /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x713fcc5c6e6f]
#35  /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x713fcc5ca0c9]
#36  0x713fcc5703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#37  0x713fce337629 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0x99
#38  0x713fcb6b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Core.so.5+0x108
#39  0x713fcbd353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Gui.so.5+0x84c
#40  0x713fcbd0b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Gui.so.5+0xac
#41  /tmp/.mount_FreeCAdODja2/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x713fc5c56bea]
#42  /tmp/.mount_FreeCAdODja2/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x713fc9d10ad7]
#43  /tmp/.mount_FreeCAdODja2/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x713fc9d13ac7]
#44  /tmp/.mount_FreeCAdODja2/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x713fc9d140ce]
#45  0x713fcb70aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Core.so.5+0x66
#46  0x713fcb6b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Core.so.5+0x11b
#47  0x713fcb6bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libQt5Core.so.5+0x82
#48  0x713fce29ab27 in Gui::Application::runApplication() from /tmp/.mount_FreeCAdODja2/usr/bin/../lib/libFreeCADGui.so+0xa27
#49  /tmp/.mount_FreeCAdODja2/usr/bin/freecad(+0x7620) [0x5b925772d620]
#50  /lib/x86_64-linux-gnu/libc.so.6(+0x28150) [0x713fcb028150]
#51  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x89) [0x713fcb028209]
#52  /tmp/.mount_FreeCAdODja2/usr/bin/freecad(+0x7981) [0x5b925772d981]
```
[csatorna_vizlopo6.zip](https://github.com/user-attachments/files/15814971/csatorna_vizlopo6.zip)

### Full version info

```shell
OS: Ubuntu 23.10 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git) AppImage
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Hungarian/Hungary (hu_HU)
Installed mods: 
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Curves 0.6.37
  * Ondsel-Lens 2024.5.6.01
  * fasteners 0.5.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-13:
@bgbsww FYI

szimszon, 2024-06-19:
It doesn't crash anymore with 

```
OS: Ubuntu 23.10 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Hungarian/Hungary (hu_HU)
Installed mods: 
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Curves 0.6.37
  * Ondsel-Lens 2024.5.6.01
  * fasteners 0.5.21

```

bgbsww, 2024-06-19:
Likely fixed by #14699.

closed 2024-07-01 by maxwxyz