FreeCAD/FreeCAD #14657: [Feature request] Sketcher: add dimensioning on the go (OVP) in Fillet and Chamfer

MiqCG, 2024-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Sketcher, with configuration Edit > Preferences > Sketcher > General > Tool Parameters with On-View-Parameters in "Only dimensional" or "All", Chamfers and Fillets don't have dimensioning on the go.

I think is very interesting for speeding workflow purposes and UI/UX coherence (with respect other tools that already have dimensioning on the go (line, arc, circle, rectangle, poligons, ellipses, slot...).

Related and derived from #12464

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git) AppImage
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * BillOfMaterials 0.0.8.5
  * fasteners 0.5.22
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/18265355/b438710e-ff44-4a7e-9ad2-9a411825edf9

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-07-31:
The problem being that current fillet tool creates the arc immediately after user select the edges.

So it would need to add one click to set the radius so that the arc follows the mouse. As it is in slot for example.

Alternatively we could automatically add a radius constraint and open the datum edit dialog. But that would be very intrusive...

Alternatively we could design a different solution than the OVP for this case. We could have a tool widget with a checkbox 'apply radius' and a spinbox where you would put the value. This way we don't have to add one click. But I don't like this too much since it would be a different UI and maybe clunky to use.

Perhaps the best solution is to add one click indeed. Do you guys know how it is handle by SW, fusion etc? @pierreporte perhaps you know? Do they have a third click?

PaddleStroke, 2025-07-31:
@chennes please assign me thanks