FreeCAD/FreeCAD #14658: [Problem] Sketcher On-View-Parameters options (Preferences): "All" concept is unclear

MiqCG, 2024-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When navigating in the Edit > Preferences > Sketcher > General > Tool parameters, there are 3 options in On-View-Parameters: Disable, Only dimensional, All.

The concept "All" seems unclear to me as there is no reference what are "all options", as there is "Only dimensional" and "Disable". So, of course, makes no sense that "All" is enable (only dimensional) and disable at the same time, and the doubt persists. Indeed the tooltip explains that "All" involves Position, but tooltip only appear if the user knows there is one or if by chance stops the pointer to the drop-down menu for a second (there is no visual indication of tooltips existence, by the way).

For faster understanding, I propose to change "All" for "Position and dimensions" or "Positioning and dimensional" or what a native English speaker consider best (sorry).

# Subproject(s) affected?
Sketcher , Preferences , UI/UX

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git) AppImage
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * BillOfMaterials 0.0.8.5
  * fasteners 0.5.22
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-16:
To add from https://github.com/FreeCAD/FreeCAD/issues/13555
When using OVP with the setting dimension only the OVP of the line tool with mode 2 points are not displayed.
When using the setting for all OVP the OVPs are also displayed for the second point.
[image]

Also, when the first point is placed a M for cycling the mode resets the tool and the first point has to be placed again. I get the behavior for all OVP but dimension only could keep the first point. I guess this part is a FR as all tools handle this the same way.

PaddleStroke, 2024-06-17:
@MiqCG the change you offer seems reasonable to me. If a native speaker can confirm the best wording to use we'll update the text.

@maxwxyz the bspline OVP PR introduce the possibility to not reset the tool when changing mode. And yes it could be introduced for some tools like the line tool.
As for the '2 points' mode I see several possibilities:
- Make an exception for this mode, show the positioning OVP of start and end point even though they are dimensional. As it's specifically a positional mode it would make sense.
- Hide this mode if the setting is not all.

Perhaps its best to keep the issues separated though?

MiqCG, 2024-06-20:
@chennes  , could you please help us with the linguistic correctness of the aforementioned text?

chennes, 2024-06-21:
I think I'd be inclined to go with:
```
None
Dimensions only
Position and dimensions
```
but let's also loop in the @FreeCAD/design-working-group here.

pierreporte, 2024-06-21:
@chennes Simple and explicit, it’s pretty good.

obelisk79, 2024-06-21:
I had to change my settings for the sanity check, I agree with the proposal from chennes.

PR #14868 (2024-06-23): Sketcher: Rename OVP preference options
fixes #14658 
renames preferences for OVP to:
- None
- Dimensions only
- Position and dimensions

@FreeCAD/design-working-group FYI
closed 2024-06-23 by chennes