FreeCAD/FreeCAD #14663: Pad up to face fails in some scenarios

chopsywa, 2024-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a tricky one and hard to replicate, so I have attached a model I created where the problem exists. This model is fine in 0.21.

The problem can be brought about by loading the model, then  changing the Binder001 in Body Cover to frozen, or detached, essentially making a change to the model. At this point Sketch 017 will  fail. There are various issues, but the one that seems most obvious is that the Pad MountPosts fails. This pad uses up to face as the type and rather than padding up to the nominated face, it appears to pad FROM the nominated face to infinity. I tried to replicate this scenario with a simpler two body model, but naturually it worked, so I included my file so whomever looks at it can see exactly what is happening to cause the issue.

I initially drew this in Onsdel, but then I updated Onsdel to the latest dev build and it has the same issue as Freecad 0.22. In Freecad 0.21 it's all good. 

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * Curves.backup1706353084.2536438 0.6.13
  * fasteners.backup1718262130.4318058 0.5.22
  * OpenDark.backup1718262281.5371034 2024.3.13
```

### Subproject(s) affected?

PartDesign

### Anything else?

[BoseQC15Headset.zip](https://github.com/user-attachments/files/15822268/BoseQC15Headset.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-06-13:
I can reproduce
```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Debug
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
```
this is a minimal fail-test:
[U2F-error.FCStd.zip](https://github.com/user-attachments/files/15827855/U2F-error.FCStd.zip)

The UpToShape mode (without selection) fails too but only for one of the two circle.
[image]

chopsywa, 2024-06-14:
I also realise looking back at my model, that every sketch has lost its attachment.

chopsywa, 2024-06-14:
It appears to be when a face is curved, but not all circumstances. I can sometimes pad up to a curved face. It seems to be easiest to trigger the event when a pad overlaps the target face and the adjacent face.

Another example.
[test-toface.zip](https://github.com/user-attachments/files/15831476/test-toface.zip)

maxwxyz, 2024-06-24:
@FlachyJoe was this fixed with #11135 ?

FlachyJoe, 2024-06-24:
@maxwxyz no

maxwxyz, 2024-08-17:
@FlachyJoe any news on this one? Will this be a feasible fix for 1.0?

chopsywa, 2024-09-17:
I just tested this with the file provided by @FlachyJoe on build 38768 and the problem persists.

FlachyJoe, 2024-09-17:
@chopsywa #16030 is not merged yet so nothing has changed.

AjinkyaDahale, 2024-09-17:
While trying to make a macro, I notice that the pipe _really_ does not like if the "spine" is created "after" the "base/profile".

AjinkyaDahale, 2024-09-17:
Macro to reproduce: 
[bug-14663.zip](https://github.com/user-attachments/files/17033605/bug-14663.zip)

PR #16030 (2024-09-19): [PD] fix pad uptoface and uptoshape
Fix #14663 
Fix #16557
closed 2024-09-19 by chennes