FreeCAD/FreeCAD #14669: After opening and closing a sketch without even touching anything the Body broke

szimszon, 2024-06-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Load the project
2. Recompute the Body object just to check the integrity
3. Open MasterSketch
[image]
4. Close the sketch
5. The AdditivePipe feature is broken
[image]
[vizlopo.zip](https://github.com/user-attachments/files/15832161/vizlopo.zip)

### Full version info

```shell
OS: Ubuntu 23.10 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git) AppImage
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Hungarian/Hungary (hu_HU)
Installed mods: 
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Curves 0.6.37
  * Ondsel-Lens 2024.5.6.01
  * fasteners 0.5.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-15:
I even get an error just opening the file. I am pretty sure the pipe would not work with such a master sketch with multiple closed overlapping profiles.

szimszon, 2024-06-16:
> I even get an error just opening the file. I am pretty sure the pipe would not work with such a master sketch with multiple closed overlapping profiles.

When I open the file I see only this messages:

```
07:18:58  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
07:18:58  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
07:18:58  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.

```
with 

```
OS: Ubuntu 23.10 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git) AppImage
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Hungarian/Hungary (hu_HU)
Installed mods: 
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Curves 0.6.37
  * Ondsel-Lens 2024.5.6.01
  * fasteners 0.5.21

```

szimszon, 2024-06-16:
> I even get an error just opening the file. I am pretty sure the pipe would not work with such a master sketch with multiple closed overlapping profiles.

The modell is created with FreeCAD weekly builds all from the beginning.

I'm not using the entire object for the path just one edge from the Master sketch:
[image]

For the profile I have a separate sketch called _OutputPipeSketch_ . It has no overlapping profiles.

The AdditivePipe works with this setup for me as you can see in this picture:
[image]

If this should not work at all then there is a second issue here as FreeCAD allows me to do it and it even works as expected.

The issue for me here is still if I open the MasterSketch then do nothing and close it the geometry references get screwed and my model broke.

If this is a result of a "not valid" MasterSketch it looks to me as a bad thing as my understanding of a master sketch is that I can draw any geometry in it without make it a valid closed and not overlapping pofiles for future references.

szimszon, 2024-06-16:
After all my question is:

* Is FreeCAD not designed to handle an edge from a Sketch where multiple open and/or overlapping profiles are present for Pipe?
  * then it should inform the user about it
* If I don't do anything within a sketch not even clicking on anything then why recompute?
* If "not valid" sketch is used as a master sketch for future external references (which is a recommended way - as far as I know the recommendations) and it has a capability to screw geometry references only by opening and closing it, do it pose a big risk with complex projects? I needed aprox. 10 minute to fix my model after opening and closing the master sketch. (Any it is repeatable - if you open and close the master sketch it will broke my model again and again)

szimszon, 2024-06-16:
Just checked with the latest weekly build:

```
OS: Ubuntu 23.10 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git) AppImage
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Hungarian/Hungary (hu_HU)
Installed mods: 
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Curves 0.6.37
  * Ondsel-Lens 2024.5.6.01
  * fasteners 0.5.21

```

* I can load the file.
* AddtitvePipe works.
* I open MasterSketch.
* I close MasterSketch.
* AdditivePipe broken.

[image]

```
08:02:19  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
08:02:19  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
08:02:19  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#DatumPlane.Support missing element reference vizlopo#Sketch ;g53;SKT;:H,E.Edge11
08:02:33  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch004.ExternalGeometry missing element reference vizlopo#Sketch ;g61;SKT;:H,E.Edge10
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch004.ExternalGeometry missing element reference vizlopo#Sketch ;g59;SKT;:H,E.Edge8
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch006.ExternalGeometry missing element reference vizlopo#Sketch ;g54;SKT.Edge47
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch006.ExternalGeometry missing element reference vizlopo#Sketch ;g41;SKT;:H,E.Edge12
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch006.ExternalGeometry missing element reference vizlopo#Sketch ;g36;SKT;:H,E.Edge14
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch006.ExternalGeometry missing element reference vizlopo#Sketch ;g19;SKT.Edge18
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch005.ExternalGeometry missing element reference vizlopo#Sketch ;g29;SKT;:H,E.Edge6
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch005.ExternalGeometry missing element reference vizlopo#Sketch ;g57;SKT.Edge50
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#AdditivePipe.Spine missing element reference vizlopo#Sketch ;g37;SKT.Edge34
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch007.ExternalGeometry missing element reference vizlopo#Sketch ;g48;SKT.Edge44
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch007.ExternalGeometry missing element reference vizlopo#Sketch ;g49;SKT.Edge45
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch007.ExternalGeometry missing element reference vizlopo#Sketch ;g50;SKT.Edge46
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch001.ExternalGeometry missing element reference vizlopo#Sketch ;g24;SKT.Edge24
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch001.ExternalGeometry missing element reference vizlopo#Sketch ;g27;SKT.Edge26
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch001.ExternalGeometry missing element reference vizlopo#Sketch ;g13;SKT.Edge3
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch001.ExternalGeometry missing element reference vizlopo#Sketch ;g31;SKT.Edge29
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch008.ExternalGeometry missing element reference vizlopo#Sketch ;g42;SKT.Edge35
08:02:33  <PropertyLinks> PropertyLinks.cpp(453): vizlopo#Sketch008.ExternalGeometry missing element reference vizlopo#Sketch ;g43;SKT;:H,E.Edge13
08:02:33  Invalid shape name ?Edge13
Clear constraints to external geometry
08:02:33  Invalid shape name ?Edge45
Clear constraints to external geometry
08:02:33  Invalid shape name ?Edge12
Clear constraints to external geometry
08:02:33  Invalid shape name ?Edge3
Clear constraints to external geometry
08:02:33  Invalid shape name ?Edge8
Clear constraints to external geometry
08:02:33  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
08:02:33  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
08:02:33  AdditivePipe: A fatal error occurred when making the pipe
08:02:33  AdditivePipe: A fatal error occurred when making the pipe
```

maxwxyz, 2024-06-16:
@bgbsww is this related to TNP changes?

szimszon, 2024-06-19:
The problem also present in version:

```
OS: Ubuntu 23.10 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Hungarian/Hungary (hu_HU)
Installed mods: 
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Curves 0.6.37
  * Ondsel-Lens 2024.5.6.01
  * fasteners 0.5.21

```

bgbsww, 2024-07-04:
This file does not open cleanly anywhere.  There are broken external references in sketches somewhere.  Not much I can really do with that, unfortunately.  Is there any chance of getting a file that does not have external geo problems in it?

szimszon, 2024-07-04:
Unfortunately I have no backup of the file without the problem :(

szimszon, 2024-07-04:
What worked for me in the past. If you correct all the refs and the body can be recomputed. Go into the sketch and close it. It will brake again.

bgbsww, 2024-07-04:
Hmm.  Any chance of a simpler example against a current build?  Or even a corrected file saved pre-sketch opening?  I don't really have anything to work with here.

szimszon, 2024-07-04:
I'll need a new build with the  "Sketch: argument not found" fix (#15117 ) as right now the situation is worse than before.

adrianinsaval, 2024-07-06:
> I'll need a new build with the "Sketch: argument not found" fix (#15117 ) as right now the situation is worse than before.

should be up now

szimszon, 2024-07-07:
The situation is better now.
[vizlopo_good.zip](https://github.com/user-attachments/files/16120576/vizlopo_good.zip)
With this fixed file if you just open and close the MasterSketch the modell is not entirely broken.

I got this errors:
[image]

```
21:54:10  Sketcher External Geometry without ExternalGeometryExtension: e2c6f381-1eb0-4a40-aa74-a078a839b302 
21:54:10  Invalid shape name ?Edge45
Clear constraints to external geometry
21:54:10  Sketcher External Geometry without ExternalGeometryExtension: d68c6b01-37a5-45b2-a353-bda0206a7254 
21:54:10  Invalid shape name ?Edge12
Clear constraints to external geometry
21:54:10  Sketcher External Geometry without ExternalGeometryExtension: 886047fe-4575-4a4e-9789-d3735057d4af 
21:54:10  Sketcher External Geometry without ExternalGeometryExtension: 20a4b520-6955-40c5-856d-479012d1ceba 
21:54:10  Sketcher External Geometry without ExternalGeometryExtension: 47471b8e-e021-496a-814b-7aded612743f 
21:54:10  Sketcher External Geometry without ExternalGeometryExtension: ce930878-13bc-4d92-ba2c-d5c75d530636 
21:54:10  Sketcher External Geometry without ExternalGeometryExtension: ba94a5c4-a131-49ae-be77-06108405f51e 
21:54:10  Invalid shape name ?Edge8
Clear constraints to external geometry
21:54:10  Sketcher External Geometry without ExternalGeometryExtension: e3932ed2-64a1-4c53-827c-158bc373455c 
21:54:10  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
21:54:10  Invalid shape name ?Edge6
Clear constraints to external geometry
```

szimszon, 2024-07-07:
After closing and opening a file I got:
```
22:07:07  Reading failed from embedded file: StringHasher.Table.txt (7540 bytes, 838 bytes compressed)
22:07:07  <ElementMap> <string>(2): No hasherRef
```

szimszon, 2024-07-07:
After recompute close open I see only this:

```
22:10:36  <ElementMap> <string>(2): No hasherRef
22:10:36  <ElementMap> <string>(2): No hasherRef
22:10:36  <ElementMap> <string>(2): No hasherRef
22:10:36  <ElementMap> <string>(2): No hasherRef
22:10:36  <ElementMap> <string>(2): No hasherRef
22:10:36  <ElementMap> <string>(2): No hasherRef
22:10:36  <ElementMap> <string>(2): No hasherRef
22:10:36  <ElementMap> <string>(2): No hasherRef
22:10:36  <ElementMap> <string>(2): No hasherRef
22:10:36  <ElementMap> <string>(2): No hasherRef
22:10:36  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
22:10:36  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
```

szimszon, 2024-07-07:
Overall it seems good now.

bgbsww, 2024-07-07:
Thank-you.  Those messages are still a problem to fix, but should have no behavioral change.  They are either not necessary, or potentially performance impacting.  Thanks for the testing and feedback!

szimszon, 2024-07-08:
uw

kinghat, 2024-07-11:
not sure if related but opening and closing the highlighted sketch i get: 

> Sketcher External Geometry without ExternalGeometryExtension: dda9c76d-2c76-4f0d-825a-7229fa15225c

[image]

> OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.38079 (Git) AppImage
> Build type: Release
> Branch: main
> Hash: de508e437d13e7eb35baffceea496fb9107888cb
> Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
> Locale: English/United States (en_US)
> Installed mods: 
>   * OpenTheme 2024.7.3

[sonoff_controller_box.zip](https://github.com/user-attachments/files/16182396/sonoff_controller_box.zip)

bgbsww, 2024-07-23:
@maxwxyz I don't think this is a blocker anymore.  There are No HasherRef noise messages to clean up that should get fixed, but no remaining functionality issues if I understand this correctly.

maxwxyz, 2024-07-29:
So what is left here or can this be closed. Is there also a fix with #15575?
@szimszon Please test again with the latest weekly tomorrow: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Maybe even try to create it from scratch if this is an option. TIA

szimszon, 2024-07-30:
Unfortunately I see only old one with the last commit 5 days ago.

[image]

maxwxyz, 2024-07-30:
@adrianinsaval the appimage did not get an update for the weekly, any idea?

bgbsww, 2024-08-09:
@szimszon Looks like there are builds now.  Any continued messages?

kinghat, 2024-08-09:
> @szimszon Looks like there are builds now. Any continued messages?

if my [issue](https://github.com/FreeCAD/FreeCAD/issues/14669#issuecomment-2223678140) in the comment above was related, i no longer see any messages fwiw.

maxwxyz, 2024-08-09:
@szimszon could you please test

PR #15845 (2024-08-12): Toponaming: Fix transformed; abstract index element name generation
fix #15515  fix #15468 fix #14669 Correct stringhasher save.
closed 2024-08-12 by chennes