FreeCAD/FreeCAD #14670: Trimex does not work when using links.

kaiwas, 2024-06-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The trimex tool cannot extrude faces from an object if it is a reference.
In this case, a regular “extrude” works great.

Video demonstration.

https://github.com/FreeCAD/FreeCAD/assets/32273833/de66c347-c6e7-433c-b391-e45a4d13d821

Test file is attached. Rename TXT --> FCstd
[Link_Trimex.txt](https://github.com/user-attachments/files/15832667/Link_Trimex.txt)

### Full version info

```shell
0.22dev
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-14:
@Roy-043 FYI, is this a bug or just not possible now?

Roy-043, 2024-06-21:
Much of the Draft code is not yet 'link-aware'. It is my intention to work on this in the next dev cycle.

kaiwas, 2024-06-21:
> Much of the Draft code is not yet 'link-aware'. It is my intention to work on this in the next dev cycle.

great news! Thank you.
Links have great potential. In freecad, links work differently than in blender, for example. there it is just a copy of the object. and in freecad it is possible to change some properties of a link and because of this they behave like autonomous objects. and that's very cool!

PR #18314 (2024-12-05): Draft: make Trimex extrude mode link-aware
Fixes #14670.
commit 6858a1c0 (2024-12-05): Draft: make Trimex extrude mode link-aware

Fixes #14670.
closed 2024-12-05 by Roy-043