FreeCAD/FreeCAD #14673: Std_Measure: Error for angle between face and edge of that face

Roy-043, 2024-06-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you measure the angle between a face and an edge of that face there is a warning. I am not saying this angle makes sense. But this situation may occur accidentally.

Test:
Create a Part_Box.
Start Std_Measure.
Select a face and an edge belonging to that face.
Result: Warning:
`10:26:48  <Gui> Document.cpp(822): Cannot update representation for Unnamed#Angle`

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-24:
@hlorus FYI

karliss, 2025-03-18:
Testing with 1.0 I am not getting a warning anymore, but I am always getting 90° . I somewhat expected always 0° (edge is parallel to plane), but I guess measuring angle against plane normal vector also makes sense.

The placement of  preview widget on the other hand was somewhat random. Sometimes it was perpendicular to the plane which would be correct if it was also parallel to the edge, but sometimes it was within plane which was completely wrong.

In somwhat recentish 1.1dev build it's impossible to select face+edge(doesn't matter if they touch or not) unless you ctrl click. Otherwise same behavior as 1.0. 90° with bad visualization. 

Examples of bad visualization:
Additional lines drawn by hand to make the 3d plane clearer.
[image]
[image]

Preferred visualization:

Ideally the visualization should be placed at the end of edge (1), but other placements (2, 3) are also acceptable as long as one side is parallel to edge and other perpendicular to face.
[image]

If it's desired to consider edge parallel to face as 0 degrees then the visualization needs to be slightly different.

pierreporte, 2026-02-08:
I tested and the issues are fixed, except those still mentioned in the previous comments. There are a few more issues:

I use the light theme, and by default the lines are now white, so almost inivisble.

The pixel sizes in the preferences are more usable but still not very realistic:

[image]

[image]

There is no recompute for style but then if you change the style in the preferences between two measure you have two different styles on screen. And the colors are hardcoded in the file because when I save and reopen the colors are the same as before saving. These colors should be part of the theme. @kadet1090 can help for this.

pierreporte, 2026-02-09:
I wrote my comment on the wrong place sorry