FreeCAD/FreeCAD #14678: Insert Horizontal Extent Dimension → segfault

mpfusion, 2024-06-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To reproduce:

- open attached file
- open TD workbench
- insert default page
- select PCB_Sketch_f277
- insert view
- click bottom arrow → OK
- select north-west and south-east edges (see image)
- click “Insert Horizontal Extent Dimension”
→ segfault
[td-segfault.tar.gz](https://github.com/user-attachments/files/15836204/td-segfault.tar.gz)
[image]

```
Geom::baseFactory - OCC error - GC_MakeCircle::Value() - no result - while making spline
Geom::baseFactory - OCC error - GC_MakeCircle::Value() - no result - while making spline
Geom::baseFactory - OCC error - GC_MakeCircle::Value() - no result - while making spline
Geom::baseFactory - OCC error - GC_MakeCircle::Value() - no result - while making spline
Geom::baseFactory - OCC error - GC_MakeCircle::Value() - no result - while making spline
Geom::baseFactory - OCC error - GC_MakeCircle::Value() - no result - while making spline
Geom::baseFactory - OCC error - GC_MakeCircle::Value() - no result - while making spline
DrawProjectSplit::scrubEdges - OCC fuse raised warning(s):
BOPAlgo_AlertAcquiredSelfIntersection
BOPAlgo_AlertAcquiredSelfIntersection

TechDraw::DrawViewDimExtent: Link(s) to object(s) 'PCB_Sketch_f277' go out of the allowed scope 'DimExtent'. Instead, the linked object(s) reside within 'Board_f277'.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7f76add94050]
#1  0x7f767196e650 in TopoDS_Builder::Add(TopoDS_Shape&, TopoDS_Shape const&) const from /lib/x86_64-linux-gnu/libTKBRep.so.7+0x20
#2  0x7f7644a650a5 in TechDraw::DrawDimHelper::minMax3d(TechDraw::DrawViewPart*, std::vector<TechDraw::ReferenceEntry, std::allocator<TechDraw::ReferenceEntry> >, int) from /home/mp/usr/bin/freecad-bin/lib/TechDraw.so+0x1e5
#3  0x7f7644b27d3c in TechDraw::DrawViewDimExtent::getPointsExtent(std::vector<TechDraw::ReferenceEntry, std::allocator<TechDraw::ReferenceEntry> >) from /home/mp/usr/bin/freecad-bin/lib/TechDraw.so+0x46c
#4  0x7f7644b28466 in TechDraw::DrawViewDimExtent::execute() from /home/mp/usr/bin/freecad-bin/lib/TechDraw.so+0x1a6
#5  0x7f76b018fd5a in App::DocumentObject::recompute() from /home/mp/usr/bin/freecad-bin/lib/libFreeCADApp.so+0xba
#6  0x7f76b012d054 in App::Document::_recomputeFeature(App::DocumentObject*) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADApp.so+0x344
#7  0x7f76b0132437 in App::Document::recomputeFeature(App::DocumentObject*, bool) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADApp.so+0xe7
#8  0x7f7644a61c25 in TechDraw::DrawDimHelper::makeExtentDim3d(TechDraw::DrawViewPart*, std::vector<TechDraw::ReferenceEntry, std::allocator<TechDraw::ReferenceEntry> >, int) from /home/mp/usr/bin/freecad-bin/lib/TechDraw.so+0x665
#9  0x7f7644a62018 in TechDraw::DrawDimHelper::makeExtentDim3d(TechDraw::DrawViewPart*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<TechDraw::ReferenceEntry, std::allocator<TechDraw::ReferenceEntry> >) from /home/mp/usr/bin/freecad-bin/lib/TechDraw.so+0xd8
#10  0x7f7644e79596 in execExtent(Gui::Command*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /home/mp/usr/bin/freecad-bin/lib/TechDrawGui.so+0x2e6
#11  0x7f7644e79a09 in CmdTechDrawHorizontalExtentDimension::activated(int) from /home/mp/usr/bin/freecad-bin/lib/TechDrawGui.so+0x129
#12  0x7f76b0c14ea1 in Gui::Command::_invoke(int, bool) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x281
#13  0x7f76b0c15343 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x133
#14  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2e8f4f) [0x7f76ae45bf4f]
#15  0x7f76aeef2782 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x42
#16  0x7f76aeef53ab in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xab
#17  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2e3b62) [0x7f76af079b62]
#18  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2eb994) [0x7f76af081994]
#19  0x7f76aef3adb8 in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1a8
#20  0x7f76aeef8fae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#21  0x7f76aef01552 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x13c2
#22  0x7f76b0bb9868 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x68
#23  0x7f76ae4246f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#24  0x7f76aeeff65e in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1be
#25  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1be025) [0x7f76aef54025]
#26  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c0f60) [0x7f76aef56f60]
#27  0x7f76aeef8fae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#28  0x7f76b0bb9868 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x68
#29  0x7f76ae4246f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#30  0x7f76ae8033ed in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x80d
#31  0x7f76ae7d7cac in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#32  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x6deca) [0x7f76a7fe4eca]
#33  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x299) [0x7f76ac1c17a9]
#34  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x54a38) [0x7f76ac1c1a38]
#35  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f76ac1c1acc]
[... 31 lines cut ...]
https://forum.freecad.org/viewtopic.php?t=88440

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Installed mods: 
  * BOLTSFC
  * kicadStepUpMod 11.1.8
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-06-16:
The immediate cause is the inclusion in the selection of the sketch and the 2d edges in the View. 
With both 2d and 3d geometry in the selection, the program gets confused between making a 3d extent dim and a 2d extent dim.

PR #14713 will protect against the failure.

mpfusion, 2024-06-18:
This sort-of works, but not quite. The segfault is gone, but now I get a popup: “Incorrect selection
Selection contains both 2d and 3d geometry” on the first try. When select the same edges a second time, it succeeds and the dimension is created.
[image]

WandererFan, 2024-07-17:
I think this is now fixed.  I can't reproduce it on recent main branch.

luzpaz, 2024-07-17:
@WandererFan Do you mind specifying what revision of main branch ?

luzpaz, 2024-08-29:
@mpfusion thanks for the follow-up. Can you open a separate ticket in that case as I close this one?

closed 2024-08-29 by luzpaz