FreeCAD/FreeCAD #14684: Incorrect file names displayed when linking

prokoudine, 2024-06-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create several files per each part in an assembly
2. Create a new file for the assembly, add the assembly object, save the file
3. Use 'Insert Component' from the toolbar to link parts
4. FreeCAD will display names of parts file as Unnamed_%count%.FCStd instead of actual file names

[image]

Closing and reopening files fixes this. Unsure if it's a core issue or assembly wb issue, up to you to decide :)

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git) AppImage
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.9.0
  * DynamicData 2.60.0
  * Assembly4 0.50.6
  * cura_engine
  * ExplodedAssembly
  * fasteners 0.5.21
  * sheetmetal 0.4.11
  * A2plus 0.4.61
  * BIM021 2021.12.0
  * OpenTheme 2024.4.20
  * Render 2023.12.28
  * Rocket 3.2.4
  * Curves 0.6.21
  * freecad.gears 1.2.0
  * Assembly3 0.12.0
  * boltsfc 2022.11.5
  * OpenDark 2023.12.17
  * dodo 1.0.0
  * Ondsel-Lens 2024.4.16.01
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-14:
I can confirm. If the file is newly created then the name displayed is still 'Unnamed'. If you close the document and re-open it then it's correct.

This is a bug in core as the code in insert component is very simple:
```
        docList = App.listDocuments().values()

        for doc in docList:
            docItem = QtGui.QTreeWidgetItem()
            docItem.setText(0, doc.Name + ".FCStd")
```
With the python console some tests confirm that the 'Name' property is not updated after saving the file:
```
d = App.ActiveDocument
>>> d.Name
'Unnamed1'
>>> d.Label
'test2'
```

We can easily fix the issue in Assembly Insert component by using the Label instead of Name. But perhaps you want to tackle to  underlying issue @wwmayer ?

krushia, 2024-06-15:
This is one of the things discussed in my sloppy Heisenbug https://github.com/FreeCAD/FreeCAD/issues/13841

PaddleStroke, 2024-06-17:
Indeed you are correct it seems to be a consequence of the bug you are describing.
For the bug in assembly I solve it by using the doc Label instead.

closed 2024-06-17 by chennes