FreeCAD/FreeCAD #14685: Assembly not on the list of workbenches on Ubuntu

FEA-eng, 2024-06-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the newest weekly build on Ubuntu, I don't have the Assembly workbench on the list. I can only access it using the template button on the start page.

[image]
[image]

It works normally on Windows.

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-07-04:
I still have this issue. I've noticed that Assembly was deactivated on the list of workbenches and now it's normally available but that makes me wonder why it's deactivated by default on Ubuntu.

maxwxyz, 2024-07-04:
Definitely if this is an issue. It seems that no one could reproduce this. Could you try it without your existing config?
Anyone else on Ubuntu who could confirm this?

FEA-eng, 2024-07-05:
> Definitely if this is an issue. It seems that no one could reproduce this. Could you try it without your existing config? Anyone else on Ubuntu who could confirm this?

My configuration on Ubuntu is quite new but I still cleaned it and now Assembly is enabled by default so I'm closing this issue.

closed 2024-07-05 by FEA-eng