FreeCAD/FreeCAD #14687: BIM: ArchSpace - Inconsistent computed Vertical Area

marcuspollio, 2024-06-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the BIM Workbench, the ArchSpace provides several computed values from its geometry, such as the horizontal area, the perimeter length ([both now](https://github.com/FreeCAD/FreeCAD/commit/73c64158785a595c2e9fa0fe17d39d25d66a509b), with 2 methods), and also the Vertical Area.
However, this third computed value is not always correct. For vertical extrusions (prismatic spaces based on a flat XY profile), there are no issues. But when ArchSpaces have inclined faces or curved features, the Vertical Area provided can be wrong or misleading.

In the example below, the first Space ("Trapezoid") has two inclined faces which are not computed in the Vertical Area. These two inclined faces are not really vertical indeed, so one could say it's correct.
The third and fourth Spaces ("Barrel_Vault" and "Sphere") have curved features, which are not vertical either.
Yet they are included in the calculation of the Vertical Area.

For consistency sake, one would expect that either all strictly vertical-only areas are computed, or all non-horizontal areas are computed. The latter would cover a wider range of situations. This is open to debate, of course =D

[image]

### Full version info

```shell
OS: GNU/Linux Ubuntu 22.04 LTS (DE:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.6.3
```

### Subproject(s) affected?

BIM/Arch

### Anything else?

[BIMSpace.zip](https://github.com/user-attachments/files/15845779/BIMSpace.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DeniseBryson, 2024-06-17:
Hi,
I can try to correct this as well. :) In my opinion vertical areas should only count vertical areas, otherwise we should call it non horizontal areas? Any other ideas?

yorikvanhavre, 2024-06-20:
Closing as solved by #14761

closed 2024-06-20 by yorikvanhavre