FreeCAD/FreeCAD #14690: Internal Assembly WB: Cylindric Langth limit also limits the revolution.

Panda631, 2024-06-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Constrain 2 parts with a cylindrical contrain. 

[image]

3. Limit the on axis sliding length.

[image]

4. The part can't longer rotated

Here the file:
[FC10_Assy.zip](https://github.com/user-attachments/files/15846665/FC10_Assy.zip)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.21
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.1.23
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-15:
I guess because the angle limit is 0 on min and max?
@PaddleStroke can you confirm?

We discussed this briefly in https://github.com/FreeCAD/FreeCAD/issues/13850#issuecomment-2098769272
I guess either we need to handle 0 better but we cannot just ignore 0. Or make a new checkbox for the rotational limits (if a joint has multiple limits) or even a checkbox for each field (min/max), so you could define a min limit but don't limit max...

@FreeCAD/design-working-group any other ideas?
@FEA-eng how is FEM handling different limits?

Panda631, 2024-06-15:
Why not handling 0  in min and max Input field as not constraint?

pierreporte, 2024-06-15:
I think it would be better to have a checkbox for each line, and remove the “Limits” one. There are cases when there can be a minimum but no maximum, and vice-versa.

FEA-eng, 2024-06-15:
> @FEA-eng how is FEM handling different limits?

Do you mean handling of Assembly joints? FEM doesn't see and use them at all. It has its own constraints defined independently.

maxwxyz, 2024-06-16:
@Panda631 0 as not constraint is not good. How could you limit a joint between 0 and 10 if 0 is ignored?
@FEA-eng no not joints but also regarding constraints. I think you can freeze or limit movement for certain directions in the FEM constraints. Can you specify a range in FEM and how is 0 in a ranged handled?

Panda631, 2024-06-16:
> @Panda631 0 as not constraint is not good. How could you limit a joint between 0 and 10 if 0 is ignored?

The Idea was 0 in min and max field. This ist the default setting and makes no sense as constraint. Just an Idea, a Check Box ist OK as well.

FEA-eng, 2024-06-16:
> Can you specify a range in FEM and how is 0 in a ranged handled?

No, you can only fix the displacement/rotation at 0 or prescribe a non-zero value. There are no such inequality/range constraints in commercial FEA software as well. To model something like this you usually have to include contact with rigid surfaces.

PaddleStroke, 2024-06-17:
I thought I have done as Panda mentioned : Ignore if both min and max are equal to 0. But apparantly no.

Checkboxes are fine. That would mean that we need 4 boolean properties, one for each limit. Unless we use special values (like 123456) to mean disable in limits but this is a bit hacky.

pierreporte, 2024-06-17:
The problem of special values is that the user must know them, while checkboxes are easily recognized, especially if coupled with disabling fields with an unchecked box.

PaddleStroke, 2024-06-17:
I meant the property could hold special value to mean empty, while the UI shows empty. But yes it's not ideal anyway. Checkboxes are better.

closed 2024-07-01 by chennes