FreeCAD/FreeCAD #14694: There is no option to bring back the first start widget after using it

FEA-eng, 2024-06-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I would really appreciate an option to bring back the first start widget after using it. I know it's supposed to be used only once but there are cases when it would be nice to be able to show it again. For example, to quickly change a few of the settings included there. Let's say that you choose some navigation style and theme but you don't like them. This way you could just change your choice in the first start widget instead of having to find those settings in preferences.

The current behavior is (from #13650):

_Once you've clicked "Done" on it, it will not reappear (unless you delete your config, or the controlling variable)._

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-01:
@FreeCAD/design-working-group any feedback for this

obelisk79, 2024-07-01:
I have a mock up for the start widget I can post later. My house Internet is down currently.

chennes, 2024-07-15:
@obelisk79 ping on this -- got an idea for me?

MisterMakerNL, 2024-07-15:
I got a request, on the start page.
Currently the theme's uses the user colors.
The color should be in the start-page.
There is even one bug where the color is black because it isn't set on first boot.
It changes blue when you trigger the settings.
Also couple of people reported the color to be too dark for the light theme.
And I cannot change this color true the cfg file because it will overwrite the user settings when you change theme's.

xtemp09, 2024-07-15:
The idea of this is to have "first-time wizard." I think, it would be better to add search bar to preferences or, at least, somehow simplify them.

obelisk79, 2024-07-16:
For 1.0 there should be a QPushButton added that shows the quick settings. After 1.0 it needs rearrangement.

maxwxyz, 2024-08-03:
is there any news on this?

qewer33, 2024-08-04:
I can possibly handle this if @chennes is busy, along with a few other new startpage issues (https://github.com/FreeCAD/FreeCAD/issues/14817 https://github.com/FreeCAD/FreeCAD/issues/15113)

FEA-eng, 2024-08-04:
> I can possibly handle this if @chennes is busy, along with a few other new startpage issues (#14817 #15113)

That would definitely be appreciated. IMO #14039 is the worst though. Then the two you've mentioned and this one here would be last when it comes to priority for me.

qewer33, 2024-08-05:
@FEA-eng this kinda related to the start page but unrelated, I'll just ask here quickly (we can open a separate issue if you agree). There being multiple different FEM examples in the examples section is kinda weird imho, can you (or someone else who does FEM) perhaps combine them into a single example?

qewer33, 2024-08-05:
> > There being multiple different FEM examples in the examples section is kinda weird imho, can you (or someone else who does FEM) perhaps combine them into a single example?
> 
> @qewer33 From https://forum.freecad.org/viewtopic.php?p=535941#p535941, this might also be relevant:
> 
> > FEM 2D and 3D new solver can be deleted. FEM 3D should be replaced by the one from FreeCAD FEM examples gui.

Alright thanks fr the link, I also opened a related forum post a while back. I'll open a new issue so we can discuss this further.

commit db3358d5 (2024-08-12): Make start page use QStackedWidget + other fixes

closes #14039, closes #14694, closes #14817, closes #15113
closed 2024-08-12 by chennes