FreeCAD/FreeCAD #14697: Crash(es) if spreadsheet contains another Spreadsheet::PropertySheet

krushia, 2024-06-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Crash when loading attached file, which was created by performing these steps:

1. New document
2. Open Spreadsheet WB
3. Add new spreadsheet
4. Type "qwerty" in A1
5. Right-click Property view > Show All
6. Right-click Property view > Add property
7. Create a new Spreadsheet:PropertySheet named MoreCells
8. Right-click Spreadsheet in Tree view > Send to Python console
9. In console, type `obj.MoreCells = obj.cells`
10. Right click Label2 > Expression
11. Enter expression `cells[<<A1>>]`
12. Save

This was done to demonstrate another bug, where changing that expression to `MoreCells[<<A1>>]` causes instant crash. However, I had to restart FreeCAD to get a clean log and found I couldn't open the file anymore :)

[morecells.log.txt](https://github.com/user-attachments/files/15853481/morecells.log.txt)
[morecells.FCStd.remove.this.file.extension.zip](https://github.com/user-attachments/files/15853495/morecells.FCStd.remove.this.file.extension.zip)

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git) AppImage
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a (Disabled)
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * dodo 1.0.1
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * Design456 0.0.1
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools (Disabled)
  * workfeature (Disabled)
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.5.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2 (Disabled)
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

There are legitimate reasons to want this to work, due to sheets having better dynamic slice support than list/tuple parameters, and aggregate functions only work on sheets too.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-06-15:
Crash can also be triggered without touching the console, by replacing these steps:
8. Right click MoreCells > Expression
9. Enter `cells`

Now typing any expression with `Morecells[<<A1>>]` will crash.

xtemp09, 2024-06-18:
I'll leave here the backtrace if there is somebody interested in this.

<details>
<summary>Backtrace
</summary>

```cpp
Process 24611 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x410)
    frame #0: 0x00007fffc415bf79 Spreadsheet.so`Spreadsheet::Cell::setContent(this=0x000055555c2e94f0, value=<unavailable>) at Cell.cpp:288:9
   285 
   286      clearException();
   287      if (value) {
-> 288          if (owner->sheet()->isRestoring()) {
   289              if (value[0] == '\0' || (value[0] == '\'' && value[1] == '\0')) {
   290                  return;
   291              }
(lldb) fr v
error: Spreadsheet.so 0x000057fc: DW_TAG_member '_M_local_buf' refers to type 0x000000000003f36a which extends beyond the bounds of 0x000057f2
(Spreadsheet::Cell *const) this = 0x000055555c2e94f0
(const char *) value = <variable not available>

(App::AtomicPropertyChangeInterface<Spreadsheet::PropertySheet>::AtomicPropertyChange) signaller = {
  mProp = 0x0000555557e303c0
}
(App::ExpressionPtr) newExpr = nullptr {
  pointer = nullptr
}
(lldb) bt
error: libFreeCADApp.so 0x000415c4: DW_TAG_member '_M_local_buf' refers to type 0x0000000000144da6 which extends beyond the bounds of 0x000415ba
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x410)
  * frame #0: 0x00007fffc415bf79 Spreadsheet.so`Spreadsheet::Cell::setContent(this=0x000055555c2e94f0, value=<unavailable>) at Cell.cpp:288:9
    frame #1: 0x00007fffc415c751 Spreadsheet.so`Spreadsheet::Cell::restore(this=0x000055555c2e94f0, reader=0x00007fffffffa940, checkAlias=false) at Cell.cpp:752:19
    frame #2: 0x00007fffc417a362 Spreadsheet.so`Spreadsheet::PropertySheet::Restore(this=0x0000555557e303c0, reader=0x00007fffffffa940) at PropertySheet.cpp:431:26
    frame #3: 0x00007ffff5fe4345 libFreeCADApp.so`App::PropertyContainer::Restore(this=0x000055555c2e1f00, reader=0x00007fffffffa940) at PropertyContainer.cpp:379:34
    frame #4: 0x00007ffff5ef005d libFreeCADApp.so`App::ExtensionContainer::Restore(this=0x000055555c2e1f00, reader=0x00007fffffffa940) at ExtensionContainer.cpp:306:36
    frame #5: 0x00007ffff5e7ba7a libFreeCADApp.so`App::Document::readObjects(this=0x0000555557e2bd40, reader=0x00007fffffffa940) at Document.cpp:1430:30
    frame #6: 0x00007ffff5e84025 libFreeCADApp.so`App::Document::Restore(this=0x0000555557e2bd40, reader=0x00007fffffffa940) at Document.cpp:1055:20
    frame #7: 0x00007ffff5e85b77 libFreeCADApp.so`App::Document::restore(this=0x0000555557e2bd40, filename=<unavailable>, delaySignal=true, objNames=size=0) at Document.cpp:2089:26
    frame #8: 0x00007ffff6063550 libFreeCADApp.so`App::Application::openDocumentPrivate(this=0x000055555571aa00, FileName=<unavailable>, propFileName=<unavailable>, label=<unavailable>, isMainDoc=true, createView=true, objNames=size=0) at Application.cpp:1016:24
    frame #9: 0x00007ffff60657df libFreeCADApp.so`App::Application::openDocuments(this=0x000055555571aa00, filenames=size=1, paths=size=0, labels=size=0, errs=size=0, createView=<unavailable>) at Application.cpp:800:47
    frame #10: 0x00007ffff6066705 libFreeCADApp.so`App::Application::openDocument(this=0x000055555571aa00, FileName=<unavailable>, createView=<unavailable>) at Application.cpp:691:79
    frame #11: 0x00007ffff608360a libFreeCADApp.so`App::Application::sOpenDocument((null)=<unavailable>, args=<unavailable>, kwd=<unavailable>) at ApplicationPy.cpp:263:46
    frame #12: 0x00007ffff5128023 libpython3.10.so.1.0`___lldb_unnamed_symbol3713 + 99
    frame #13: 0x00007ffff50dfadc libpython3.10.so.1.0`_PyObject_MakeTpCall + 140
    frame #14: 0x00007ffff507ba1c libpython3.10.so.1.0`_PyEval_EvalFrameDefault + 40380
    frame #15: 0x00007ffff51c23af libpython3.10.so.1.0`___lldb_unnamed_symbol4418 + 1647
    frame #16: 0x00007ffff51bd3de libpython3.10.so.1.0`PyEval_EvalCode + 190
    frame #17: 0x00007ffff520c4cd libpython3.10.so.1.0`___lldb_unnamed_symbol4668 + 157
    frame #18: 0x00007ffff520d4f9 libpython3.10.so.1.0`PyRun_StringFlags + 121
    frame #19: 0x00007ffff578dad1 libFreeCADBase.so`Base::InterpreterSingleton::runString[abi:cxx11](this=<unavailable>, sCmd="FreeCAD.openDocument('/home/xtemp09/Downloads/iss-14697-sp/morecells.FCStd')") at Interpreter.cpp:249:15
    frame #20: 0x00007ffff6d5adc6 libFreeCADGui.so`Gui::Command::_runCommand(file="/home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp", line=597, eType=App, sCmd="FreeCAD.openDocument('/home/xtemp09/Downloads/iss-14697-sp/morecells.FCStd')") at Command.cpp:725:38
    frame #21: 0x00007ffff6d5af84 libFreeCADGui.so`Gui::Command::_doCommand(file="/home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp", line=597, eType=App, sCmd=<unavailable>) at Command.cpp:677:16
    frame #22: 0x00007ffff6c70cf6 libFreeCADGui.so`Gui::Application::open(this=<unavailable>, FileName=<unavailable>, Module=<unavailable>) at Application.cpp:597:30
    frame #23: 0x00007ffff6d6c366 libFreeCADGui.so`StdCmdOpen::activated(this=<unavailable>, iMsg=<unavailable>) at CommandDoc.cpp:158:38
    frame #24: 0x00007ffff6d605ec libFreeCADGui.so`Gui::Command::_invoke(this=0x0000555555b5ff10, id=0, disablelog=<unavailable>) at Command.cpp:444:26
    frame #25: 0x00007ffff6d60c5c libFreeCADGui.so`Gui::Command::invoke(this=0x0000555555b5ff10, i=0, trigger=TriggerAction) at Command.cpp:408:12
[... cut ...]
    frame #72: 0x00007ffff3229e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:107:1), argc=1, argv=0x00007fffffffdd58, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd48) at libc-start.c:392:3
    frame #73: 0x0000555555569c05 FreeCAD`_start + 37
(lldb)
```

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37820 (Git)
Build type: Debug
Branch: main
Hash: a36a60d29dfe376c42576bbf4ed7c3fc7e9850a8
Python 3.10.12, Qt 5.15.14, Coin 4.0.2, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

closed 2025-06-05 by chennes