FreeCAD/FreeCAD #14700: Dependency graph view broken (regression)

krushia, 2024-06-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Dependency graph view is broken. Export still works though.

Last working build:
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2

First broken build:
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git) AppImage
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2

[image]

### Full version info

```shell
^
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-03:
Did libboost-devel get updated from 1.84 to 1.85 after May 17th on Conda weekly builds, and if so, could we try pinning it back to 1.84?

krushia, 2024-07-04:
I do notice that a directory "_boost" is present in the working appimage, but not in the latest appimage.

adrianinsaval, 2024-07-04:
> Did libboost-devel get updated from 1.84 to 1.85 after May 17th on Conda weekly builds, and if so, could we try pinning it back to 1.84?

no, we are still on boost 1.82

I cannot reproduce this error though

Syres916, 2024-07-05:
While I can't reproduce the error locally with a fresh Conda build environment, using the `packages.txt` from 37302 AppImage that was the last one I have that worked and the new build environment, it's clear that `graphviz` got upgraded from 9.0.0 to 11.0.0 maybe via 10.0.0. I'm struggling to see why we couldn't at least for a week try setting the packages in `create_bundle.sh` to:

```
packages="freecad=*.pre occt vtk python=3.11 blas=*=openblas numpy graphviz=9.0.0\
          matplotlib-base scipy sympy pandas six pyyaml pycollada lxml \
          xlutils olefile requests blinker opencv nine docutils \
          opencamlib calculix ifcopenshell lark appimage-updater-bridge"
```

Is it worth trying?

adrianinsaval, 2024-07-05:
huh, it seems freecad can't even find graphviz from the appimage. I removed my local install of graphviz and now when trying to see the dependency graph it tells me it can't find the executable so before it was just using the one from my system. It seems to be the executable is only searched at `/usr/bin` on linux instead of just in the PATH as it should:
https://github.com/FreeCAD/FreeCAD/blob/561e521817034ea7660de987d8ab7183ad583595/src/Gui/GraphvizView.cpp#L286-L290

This is not good for the appimage as we can't properly configure the path for the bundled binary as it would change on every launch. On the other hand even when giving freecad the path to the appimage bin directory it doesn't work, but it looks like this is because the `unflatten` binary is not being shipped in the appimage.

When disabling unflatten it works
https://github.com/FreeCAD/FreeCAD/blob/561e521817034ea7660de987d8ab7183ad583595/src/Gui/GraphvizView.cpp#L420-L421