FreeCAD/FreeCAD #14701: assembly It's not easy to find the joints

Vincenzoferrara, 2024-06-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have an assembly project, which has 50 joints, but when I have to modify a joint it is not simple.

the joints are numbered, my solutions are 2:
- rename the name of the joint with the name of the first piece example: joint_frontal_panel
- or to simplify it a bit for the user, show the joints and not the bodies, and double click on it to edit it (see assembly4)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git) AppImage
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT) [ OS: Italian/Italy (it_IT) ]
Installed mods: 
  * AutoOpenSpreadsheet
  * ProDarkThemePreferencePack 1.0.0
  * CommandPanel
  * IconThemes
  * fasteners.backup1715436808.37841 0.5.16 (Disabled)
  * FeedsAndSpeeds 0.5.0
  * ExplodedAssembly
  * OpenTheme 2024.5.16
  * Freecad-Built-in-themes-beta 1.2.2
  * fasteners 0.5.21
  * TabBar
  * Assembly4 0.50.12
  * CubeMenu
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-16:
You can rename all the joints. Auto naming is not ideal as you can change the part name and the references in the joint later.
You can currently hide all objects and show the joints but you cannot edit them by double click on the joint in the 3D view (only in the tree view).
I've changed the title for the second issue.
Otherwise it's also related to https://github.com/FreeCAD/FreeCAD/issues/12596

PaddleStroke, 2024-06-17:
I do not know if it is possible to define the double click in 3d view. Apparantly this double click is used to select the whole object.

@wwmayer Can you please advise if it is possible?

wwmayer, 2024-06-17:
No, it's not possible.

PaddleStroke, 2024-06-17:
Ah actually I have done somthing like this in Sketcher : https://github.com/FreeCAD/FreeCAD/pull/11925
Where clicks are handled specifically by VPSketch which made it possible to handle double click events.

Perhaps the same can be done in assembly when it is in edit. So not straight forward but potentially possible.

pierreporte, 2024-06-17:
Double clicks in 3D views has been also discussed in issue #11429.

PaddleStroke, 2024-07-08:
@wwmayer I was working on implementing this for joints but someone said me there's also a feature request to define double clicks for Sketches.
Would it be possible to define a 3dview double click function in ViewProvider class?
For instance it could be : 

```
bool ViewProvider::doubleClickedIn3d() {
     return false;
}
```

Then whereever the view clicks are defined (I couldn't find where that is) do some thing like : 
```
if (double clicked) {
      if (!obj->doubleClickedIn3d) {
            // Do the current select parent thing.
```

With the use of a timer maybe to separate double clicks from click then clicking 5 seconds later.

PaddleStroke, 2024-07-22:
@wwmayer ping

closed 2024-07-24 by chennes