FreeCAD/FreeCAD #14702: Applying extrude results in new extrude being created

nazar-pc, 2024-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was trying to extrude a sketch in Part workbench and clicked "Apply", expecting to see the preview of the result (because it doesn't show preview otherwise for some reason).

However I later noticed that regardless of whether I click "Ok" or "Apply" new Extrude is being created.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git) AppImage
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
```

### Subproject(s) affected?

Part

### Anything else?

I was trying to extrude the linked sketch in the part here: [Core.zip](https://github.com/user-attachments/files/15855822/Core.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-16:
Apply should create a new one so this is intended.

maxwxyz, 2024-06-16:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/13120

closed 2024-06-16 by maxwxyz