FreeCAD/FreeCAD #14704: Creating link by dragging causes unexpected errors

nazar-pc, 2024-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a standalone sketch that I want to link in a body.

I expected that this can be done by dragging original sketch with Alt and dropping onto the body, however I've got this instead:
```
04:33:47  2722.77 Tree.cpp(2571): [ComboView] ignore replace operation without parent
```

If I aim very carefully and drop link in between other elements of the body, the link is being created successfully, but turns out that the sketch that was inside of the body prior to this gets pushed out of the body for some reason, which I didn't expect at all.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git) AppImage
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-16:
Related https://github.com/FreeCAD/FreeCAD/issues/11676
@PaddleStroke FYI

FEA-eng, 2025-07-17:
Still relevant

PaddleStroke, 2025-11-17:
no