FreeCAD/FreeCAD #14706: Assembly: Joint cannot differentiate multiple identical parts

Atmelfan, 2024-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Working with moderately complex assembly. When trying to create a fixed joint (see highlighted in screenshot below) between two edges it apparently gets confused and joins it with another identical part (se second screenshot, note the original edge is still highlighted).

[image]

[image]

I included my entire workspace zipped below.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of Ondsel: 64-bit
Version: 2024.2.1.37226 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.1)
Hash: 1a1bd7290494aef925a026c87b808ac58d53c3b0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01
```

### Subproject(s) affected?

Assembly

### Anything else?

[archived.zip](https://github.com/user-attachments/files/15856074/archived.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-16:
I don't understand the issue. Is it hard to select the right feature for the joint or what is the bug? Please add steps how to reproduce.

Atmelfan, 2024-06-16:
Steps to reproduce:

    Open the workspace attached below. I don't know how to replicate the issue in a different setup.
    Select Edge A on the PCB (Body 1, highlighted in green in screenshot 1 above)
    Select Edge B on a standoff (Body 2, highlighted in green in screenshot 1 above)
    Press the "Fixed" joint button to create a joint.
    Joint is created between Edge A on Body 1 and Edge B but on Body 3 instead of Body 2 (See screenshot 2). Where Body 3 is another identical part to Body 2 (not the one selected in step 2) elsewhere in the assembly.

The expcted result would be a joint between Edge A on Body 1 and Edge B on Body 2.

Not sure how to get the dev version but I'll try.

Atmelfan, 2024-06-16:
Screenshots of above steps in dev version:

### Select _Edge B_ on _Body 2_
[image]

### Select _Edge A_ on _Body 1_. Both highlighted.
[image]

### Press the Fixed joint button.
[image]

### Huh!? Where did it go? (Note that _Edge B_ on _Body 2_ is still highlighted)
[image]

### Attached to _Edge B_ on _Body 3_ 
[image]

### Overview
[image]

### Dev version
```
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git) AppImage
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

maxwxyz, 2024-06-16:
I can reproduce now.
How did you insert the parts, they appear to be a link from a link:
[image]

@PaddleStroke FYI

Atmelfan, 2024-06-16:
I pressed "Insert component" button and selected it from the dialogue.

PaddleStroke, 2024-06-17:
Your project is overly complex. Please next time simplify it as much as possible while keeping the issue reproducible.
Enclosed the simplified file for reference.
[archived.zip](https://github.com/user-attachments/files/15871924/archived.zip)

So the architecture of your project is quite complex and a bit weird : 
[image]

Why are you putting the assemblies inside parts? This is what is messing things up as you can test in the simplified main_body file: it doesn't work properly.

PaddleStroke, 2024-06-17:
Ok so simplifying a bit more and now I can see more clearly the issue.
[archived.zip](https://github.com/user-attachments/files/15872020/archived.zip)

There are actually several issues raised by your project : 
1 - If an assembly is inside a App::Part then it doesn't work
2 - If a sub-assembly has several links to the same part, then the joint creation is using the first instance. It is not finding the correct placement.

Atmelfan, 2024-06-17:
```Why are you putting the assemblies inside parts?````
Idk, since they will be given a part-number and referenced to in BOMs as a single part I thought it made sense to put them inside a part.   
It also made a lot less klutter when trying inserting them into other assemblies since they would show up when "Only show parts" is selected (ideally I would want a show only Parts and Assemblies checkbox).

maxwxyz, 2024-07-01:
I will test if this is also an issue in Assembly2

maxwxyz, 2024-07-02:
Also buggy with Asm3

PaddleStroke, 2024-07-02:
A broader issue of the 'cannot differentiate multiple identical parts' has been opened for discussion : https://github.com/FreeCAD/FreeCAD/issues/15090

closed 2024-07-29 by yorikvanhavre