FreeCAD/FreeCAD #14709: Sketcher: Solver cannot handle elliptical arcs with equal minor and major radii

Roy-043, 2024-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

An elliptical arc with (almost) equal minor and major radii cannot be handled by the solver. There is a "Solver failed to converge" message. The problem does not occur for full ellipses.

Steps:
1. Create an elliptical arc.
2. Apply an equal constraint to its axis.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=88474

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-06-16:
Hi !
Seems fixed
[image]
```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37732 (Git)
Build type: Debug
Branch: main
Hash: 2ad397e18e754aeb61b547507a35a039ca9bf2ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
```

I could reproduce in 0.22.0dev.37709 so a recent change fixed this.

Roy-043, 2024-06-22:
This has indeed been fixed.

FEA-eng, 2024-06-22:
I still have this issue (also in 0.21.2), at least for a certain span of the arc:

[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

Roy-043, 2024-06-25:
Confirmed. There are also errors when you try to drag an endpoint of the arc:
```
12:38:12  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to move point with the id and type: (0, 1)
12:38:12  Unnamed1#Sketch: App.getDocument('Unnamed').getObject('Sketch').movePoint(0,1,App.Vector(-18.560385,-21.730375,0),0)
12:38:12  ViewProviderSketch: Drag point: Not able to move point with the id and type: (0, 1)
```

FEA-eng, 2025-07-05:
Still relevant