FreeCAD/FreeCAD #14711: Problem with partialy selected sketch

Kuzma30, 2024-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

How to reproduce
1. Select external outline of Sketch001.
[image]
2. Make Pad
3. Save and exit. No error shown in tree
4. Open file again.
5. Now it shown error
[image]
[test.FCStd.gz](https://github.com/user-attachments/files/15857341/test.FCStd.gz)

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Debug
Branch: master
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1.dev
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * sheetmetal 0.4.18
  * fasteners 0.5.22
  * OpenTheme 2024.5.16
  * Assembly3 0.12.2
  * addFC 0.3.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-16:
Ideally you use one profile to extrude in a sketch, not several connected closed loops.
@bgbsww could this be TNP related or just a bad sketch?

maxwxyz, 2024-06-16:
The error is the same as in https://github.com/FreeCAD/FreeCAD/issues/14602

Kuzma30, 2024-06-16:
Its not external geometry

Kuzma30, 2024-06-17:
May be will be useful.
Original sketch has 3 connected line.
[image]

When apply Pad command this lines converted to one

[image]

bgbsww, 2024-06-17:
This could be related to the TNP changes to cleanup Wire creation that @CalligaroV was working on, which would make it tangentially related.  However, I'm at a loss as to what the padded behavior of such a sketch would be other than undefined, and thus not really a bug to address.

Maybe there is an expected behavior here that I don't understand?  Squashing that outer wire makes sense to me, and then at that point any cleanup/fix operation is liable to wipe out the insides.  I don't know where that's happening, though.

Kuzma30, 2024-06-17:
The goal was to create the following detail from the master sketch
[image]

bgbsww, 2024-08-11:
@maxwxyz  this isn't a Topic: Toponaming.  It's a bad sketch.  It's interesting from a TNP perspective, but there's nothing to do.

maxwxyz, 2024-08-11:
ok I'll close then

closed 2024-08-11 by maxwxyz

Kuzma30, 2024-08-11:
> @maxwxyz this isn't a Topic: Toponaming. It's a bad sketch. It's interesting from a TNP perspective, but there's nothing to do.

What is a "good" sketch for https://github.com/FreeCAD/FreeCAD/issues/14711#issuecomment-2174243583?