FreeCAD/FreeCAD #14715: assembly include grid

Vincenzoferrara, 2024-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm building a cnc assembly, and sometimes I've found that I've made some mistakes in the angles so that my cnc is no longer flat.

having a grid like in part design would have allowed me to immediately discover if it wasn't square once assembled

### Full version info

```shell
beta version
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-16:
what should the grid do? Is it just visual like in the Draft WB?

Vincenzoferrara, 2024-06-17:
Yes, Just to visualize the plan, just like in dtaft, if it were possible I would also like it in part design

luzpaz, 2024-07-27:
Full [About](https://wiki.freecad.org/About) info please