FreeCAD/FreeCAD #14720: Easy to demonstrate toponame issue: adding a hole and a fillet to a base rectangle

dmgualb, 2024-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Very simple basic object. If I add a hole and later a fillet to the base sketch, it triggers a toponame problem. But if I do exactly the inverse: Add first the fillet and then the hole, everything works!

https://github.com/FreeCAD/FreeCAD/assets/113334344/07cdf8ac-73ce-49e7-a092-e6e4b7010df1

OBS: If I simultaneously add both the hole and the fillet, it won't work either. 

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * A2plus 0.4.64
  * Curves 0.6.8
  * Help 1.0.3
```

### Subproject(s) affected?

PartDesign

### Anything else?

[TestToponame22-01.zip](https://github.com/user-attachments/files/15859270/TestToponame22-01.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-17:
@bgbsww FYI
Is this the same as https://github.com/FreeCAD/FreeCAD/issues/14720 ?

Jookia, 2024-06-18:
> @bgbsww FYI Is this the same as #14720 ?

That's this issue?

maxwxyz, 2024-06-18:
sorry, meant this one: https://github.com/FreeCAD/FreeCAD/issues/14602
@bgbsww @Jookia

dmgualb, 2024-06-19:
> sorry, meant this one: #14602 @bgbsww @Jookia

I don't think it is the same, as the error here pops right away when we try to edit the base Sketch, it doesn't require saving/reopening the file.

bgbsww, 2024-07-04:
This is completely accurate - this does cause a toponaming failure.  However, the same failure appears to occur under LS3, which means that I don't think this is a blocker or 1.0 item.

It should stay open and a fix attempted, but that's a new research project, not a parity item.

I duplicated this on the snap release below; happy to have someone show a place it works.

[code]
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2024.409.0.38991 (Git) Snap 132
Build type: Release
Branch: tag: 20240407stable
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.10.12, Qt 5.15.10, Coin 4.0.1rt, Vtk 7.1.1, OCC 7.7.2
Locale: English/United States (en_US)
[/code]

bgbsww, 2024-07-04:
A couple of notes:  The console warning is `21:08:08  <App> PropertyLinks.cpp(439): Unnamed#Sketch001.Support missing element reference Unnamed#Pad ;#d;:G0;XTR;:H377:8,F.Face7` and the object elementmap looks like this:
```

>>> App.ActiveDocument.Pad.Shape.ElementReverseMap
{'Edge16': '#46;:G;XTR;:H377:7,E', 'Face6': '#47;:G;XTR;:H377:7,F', 'Edge18': '#48;:U;XTR;:H377:7,E', 'Vertex12': '#49;:U;XTR;:H377:7,V', 'Face3': '#4d;:G;XTR;:H377:7,F', 'Face7': '#4d;FAC;:H377:4,F', 'Edge1': '#4f;:G;XTR;:H377:7,E', 'Edge2': '#50;:G;XTR;:H377:7,E', 'Edge5': '#51;:G;XTR;:H377:7,E', 'Edge8': '#52;:G;XTR;:H377:7,E', 'Edge11': '#53;:G;XTR;:H377:7,E', 'Face1': '#54;:G;XTR;:H377:7,F', 'Face2': '#55;:G;XTR;:H377:7,F', 'Face4': '#56;:G;XTR;:H377:7,F', 'Face5': '#57;:G;XTR;:H377:7,F', 'Face8': '#58;:G0;XTR;:H377:8,F', 'Edge4': '#59;:U;XTR;:H377:7,E', 'Edge7': '#5a;:U;XTR;:H377:7,E', 'Edge10': '#5b;:U;XTR;:H377:7,E', 'Edge13': '#5c;:U;XTR;:H377:7,E', 'Edge15': '#5d;:U;XTR;:H377:7,E', 'Vertex2': '#5e;:U;XTR;:H377:7,V', 'Vertex4': '#5f;:U;XTR;:H377:7,V', 'Vertex6': '#60;:U;XTR;:H377:7,V', 'Vertex8': '#61;:U;XTR;:H377:7,V', 'Vertex10': '#62;:U;XTR;:H377:7,V', 'Edge9': 'g1;SKT;:H,E;:H377:5,E', 'Vertex5': ['g1v1;SKT;:H,V;:H377:5,V', 'g4v2;SKT;:H,V;:H377:5,V'], 'Vertex7': ['g1v2;SKT;:H,V;:H377:5,V', 'g2v1;SKT;:H,V;:H377:5,V'], 'Edge12': 'g2;SKT;:H,E;:H377:5,E', 'Vertex9': ['g2v2;SKT;:H,V;:H377:5,V', 'g6v2;SKT;:H,V;:H377:5,V'], 'Edge3': 'g3;SKT;:H,E;:H377:5,E', 'Vertex1': ['g3v1;SKT;:H,V;:H377:5,V', 'g6v1;SKT;:H,V;:H377:5,V'], 'Vertex3': ['g3v2;SKT;:H,V;:H377:5,V', 'g4v1;SKT;:H,V;:H377:5,V'], 'Edge6': 'g4;SKT;:H,E;:H377:5,E', 'Edge17': 'g5;SKT;:H377,E', 'Vertex11': 'g5v1;SKT;:H377,V', 'Edge14': 'g6;SKT;:H,E;:H377:5,E'}
>>> App.ActiveDocument.Pad.Shape.ElementReverseMap["Face7"]
'#4d;FAC;:H377:4,F'
>>> App.ActiveDocument.Pad.Shape.ElementReverseMap["Face8"]
'#58;:G0;XTR;:H377:8,F'
>>> App.ActiveDocument.Pad.Shape.ElementReverseMap["Face9"]
Traceback (most recent call last):
  File "<input>", line 1, in <module>
KeyError: 'Face9'
>>> App.ActiveDocument.Pad.Shape.ElementReverseMap["Face6"]
'#47;:G;XTR;:H377:7,F'
>>> 
>>> len(App.ActiveDocument.Pad.Shape.ElementReverseMap)
38
>>> App.ActiveDocument.Pad.Shape.ElementMapSize
43
>>> App.ActiveDocument.Pad.Shape.ElementMap
{'#46;:G;XTR;:H377:7,E': 'Edge16', '#47;:G;XTR;:H377:7,F': 'Face6', '#48;:U;XTR;:H377:7,E': 'Edge18', '#49;:U;XTR;:H377:7,V': 'Vertex12', '#4d;:G;XTR;:H377:7,F': 'Face3', '#4d;FAC;:H377:4,F': 'Face7', '#4f;:G;XTR;:H377:7,E': 'Edge1', '#50;:G;XTR;:H377:7,E': 'Edge2', '#51;:G;XTR;:H377:7,E': 'Edge5', '#52;:G;XTR;:H377:7,E': 'Edge8', '#53;:G;XTR;:H377:7,E': 'Edge11', '#54;:G;XTR;:H377:7,F': 'Face1', '#55;:G;XTR;:H377:7,F': 'Face2', '#56;:G;XTR;:H377:7,F': 'Face4', '#57;:G;XTR;:H377:7,F': 'Face5', '#58;:G0;XTR;:H377:8,F': 'Face8', '#59;:U;XTR;:H377:7,E': 'Edge4', '#5a;:U;XTR;:H377:7,E': 'Edge7', '#5b;:U;XTR;:H377:7,E': 'Edge10', '#5c;:U;XTR;:H377:7,E': 'Edge13', '#5d;:U;XTR;:H377:7,E': 'Edge15', '#5e;:U;XTR;:H377:7,V': 'Vertex2', '#5f;:U;XTR;:H377:7,V': 'Vertex4', '#60;:U;XTR;:H377:7,V': 'Vertex6', '#61;:U;XTR;:H377:7,V': 'Vertex8', '#62;:U;XTR;:H377:7,V': 'Vertex10', 'g1;SKT;:H,E;:H377:5,E': 'Edge9', 'g1v1;SKT;:H,V;:H377:5,V': 'Vertex5', 'g1v2;SKT;:H,V;:H377:5,V': 'Vertex7', 'g2;SKT;:H,E;:H377:5,E': 'Edge12', 'g2v1;SKT;:H,V;:H377:5,V': 'Vertex7', 'g2v2;SKT;:H,V;:H377:5,V': 'Vertex9', 'g3;SKT;:H,E;:H377:5,E': 'Edge3', 'g3v1;SKT;:H,V;:H377:5,V': 'Vertex1', 'g3v2;SKT;:H,V;:H377:5,V': 'Vertex3', 'g4;SKT;:H,E;:H377:5,E': 'Edge6', 'g4v1;SKT;:H,V;:H377:5,V': 'Vertex3', 'g4v2;SKT;:H,V;:H377:5,V': 'Vertex5', 'g5;SKT;:H377,E': 'Edge17', 'g5v1;SKT;:H377,V': 'Vertex11', 'g6;SKT;:H,E;:H377:5,E': 'Edge14', 'g6v1;SKT;:H,V;:H377:5,V': 'Vertex1', 'g6v2;SKT;:H,V;:H377:5,V': 'Vertex9'}
>>> ```

bgbsww, 2024-08-16:
Ultimately, this comes down to editing a sketch in a way that changes the geometry order, resulting in different element names.  With different element names, the attempt to match on the attachment fails after the faces rename themselves in the pad.

This category of bug can be solved either by ensuring that the geometry ids (order) never change [hard, unproven], or by adding another heuristic that tries to match faces by plane (as I've mentioned elsewhere).

bgbsww, 2024-09-27:
#16803

closed 2025-09-22 by chennes